raspiblitz/raspiblitz #1881: Electrum not connecting...

RC-RPB, 2020-12-17:
[https://termbin.com/slupy](url)

Caveat: I don't know much. 

Was connecting fine after patching 1.6.1. I remember last time that just reinstalling Electrs didn't work, I had to delete the index and start over from scratch. Updated to 1.6.2 and it's no longer working. I tried everything I know how to do, which is very little. Reinstalled Electrs, then deleted the index and reinstalled, disabled Tor, then deleted the index and reinstalled again. Still no luck.

hermanlim, 2020-12-19:
There was a comment which sounded similar to yours awhile back if I remember correctly. I believe the problem had something to do with the security certificates.

RC-RPB, 2020-12-19:
That did it. Thank you sir!

closed 2020-12-19 by RC-RPB