raspiblitz/raspiblitz #1884: Funding confirmation taking ages

maaxxyy, 2020-12-18:
Hi guys how long does funding confirmation normally take on the raspiblitz? My funds are still not confirmed from my hardware wallet. Thanks

NilByte, 2020-12-19:
It depends on which fee rate you used and the state of the Bitcoin mempool.

https://jochen-hoenicke.de/queue/#0,1w

Check the transaction status in a block explorer.

https://blockstream.info/

rootzoll, 2020-12-19:
On https://whatthefee.io you can see your chances that your transaction gets confirmed based in the fee of your transaction.

maaxxyy, 2020-12-19:
My payment is still being sent, i sent it on the 17th of Dec, with low network fee. On ledger its showing as not confirmed. Thanks for the replies but still confused lol

NilByte, 2020-12-19:
You hit a spike in the mempool. This is what happens when the price goes crazy like it is right now.

nyxnor, 2021-04-18:
@maaxxyy Solved? Also, this depends on the fees you used, not on the Blitz. Please close this issue if done.

nyxnor, 2021-05-17:
@rootzoll Please close this.

closed 2021-05-17 by rootzoll