raspiblitz/raspiblitz #1886: Running without lcd and without hdmi

sanitariu, 2020-12-20:
Hello,
How i can run raspiblitz without LCD and without HDMI ? I do not have any kind of monitor connected and i would not connect. Only ssh access.

NilByte, 2020-12-20:
You do not need to have any screen connected to the raspiblitz.

You will get the information needed through the ssh connection.

sanitariu, 2020-12-20:
I meant to disable all hdmi/lcd modules even remove with rmmod  and even delete pi user.
All for power saving and security.

NilByte, 2020-12-20:
You could try this.

https://www.raspberrypi.org/forums/viewtopic.php?t=255930#:~:text=Set%20hdmi_ignore_hotplug%3D1%20in%20%2Fboot,Pi%204%20support%20was%20merged.

And this.

https://github.com/rootzoll/raspiblitz/issues/1564

nyxnor, 2021-04-18:
@sanitariu Please close this issue.
To disble lcd and hdmi, you gotta build_sdcard.sh and use the option `headless` so it wont be added 
https://github.com/rootzoll/raspiblitz/blob/3032410b76b0da248aadb358ab21b7720db3c222/build_sdcard.sh#L83
https://github.com/rootzoll/raspiblitz/blob/3032410b76b0da248aadb358ab21b7720db3c222/build_sdcard.sh#L1079

closed 2021-04-18 by sanitariu