raspiblitz/raspiblitz #1887: Recommended LCD-Display not available

TheSaltyJ, 2020-12-20:
The recommended LCD-Display is not available at amazon.de. Would that https://www.amazon.de/gp/product/B01N48NOXI/ref=ox_sc_act_title_1?smid=A3U321I9X7C9XA&psc=1 be an suitable alternative?

rootzoll, 2020-12-21:
@Animalmother9000 any updates on the shopping list links?

Animalmother9000, 2020-12-21:
@TheSaltyJ sorry but I can't reproduce the error. Just tried it and it is available.

TheSaltyJ, 2020-12-21:
oh wow, they must have restocked it literally today! Thanks though. For the future, would that product in my comment be compatible with the setup? Feel free to close the issue.

josephtyler, 2021-03-06:
For what it's worth, I had the same issue a few weeks ago. I clicked the link and it was unavailable, so I bought another similar display (and I also tried the official Raspberry Pi 7" display) to no avail. I checked again today and it was available so I ordered it!

Do you know what would be required to support the official 7" display?

nyxnor, 2021-04-18:
@josephtyler Depends on the brand, the manufacturer normally has a custom script for the display.
@TheSaltyJ Can you close this issue?

closed 2021-04-20 by TheSaltyJ