raspiblitz/raspiblitz #1888: Use Tor to call for emergency kickstart Nodes

rootzoll, 2020-12-20:
see: https://github.com/rootzoll/raspiblitz/issues/1222#issuecomment-748678155

rootzoll, 2020-12-23:
Got info that calling bitcnodes.io thru Tor should be allowed now ... trying that out for v1.6.3

rootzoll, 2020-12-23:
Hmm ... calling the API works thru the Tor borwser but from the RaspiBlitz it still gives me the CLoud FlareCaptcha. looks like that some Tor Exit nodes are flagged others are not. Pushing again to v1.7

rootzoll, 2021-03-25:
The API is still blocked to reach thru TOR ... so the work around would be to call the API on build script with clearnet and put it into the `/home/admin` folder. That way on every release there will be a fresh local cache of quite a thousand nodes addresses - not optimal, but as a fallback not calling a central point of failure from every blitz running.

rootzoll, 2021-03-27:
One last thingto check: Check is the backup list works if sd card is build behind TOR and maybe check and provide a static fallback from assests for that case.

rootzoll, 2021-04-03:
OK - added a static fallback & the dynamic update on sd card build is only happening when fatpack option is on.

Issue should be ready fpr v1.7 release - changing back to fnal testing.

nyxnor, 2021-04-07:
This is kinda hard cause depends on the exit node. Keep an eye if the Bitnodes site maintainer facilitate the process https://github(dot)com/ayeowch/bitnodes/issues/49 (Broke the link to not ping there).

If Tor is blocked, [Tor -> X](https://gitlab.torproject.org/legacy/trac/-/wikis/doc/TorPlusVPN#you-tor-x) is the way to access a site that blocks Tor traffic
* You -> Tor -> VPN/SSH 
* You -> Tor -> Proxy

The project dont give much info on how to accomplish, in fact they even discourage it, cause it depends on proper implementation to improve privacy. Repeatdly saying `dangerous` on the text, but great part of their wiki is outdated and the link above is legacy, so uppon you to decide if it is worth it or not.

* Easiest is `VPN`, but require the user having an account. Here is the [Mullvad VPN option.](https://mullvad.net/en/help/tor-and-mullvad-vpn/) (Just have to skip the browser parts and change port 9150 to 9050. If using directly Tor>Bitnodes, the Tor exit node would have some data, now with Tor>Vpn>Exit node, only the VPN node will have that data, and will be able to acquire the info from the site. In the following days I will be testing this out. Bad part is that this cannot be default on Blitz, but maybe an option, as it is required a VPN payment.
* Tor -> `SSH`, I do not have info to provide for this now.
* More advanced and versatile is with `Proxy`. Need to use _curl_ with [Transparent Proxy](https://gitlab.torproject.org/legacy/trac/-/wikis/doc/TransparentProxy). Bad part is that it is a whole lot of changes, good part that if it works, can be set widely for everyone.

rootzoll, 2021-04-08:
I think the fallback to static file in the RaspiBlitz assets already sloves this good enough for now ... its a 3MB text file with lots of nodes to try - some will still be available to reconnect the bitcoin node to the network: https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/home.admin/assets/fallback.nodes

closed 2021-04-18 by rootzoll