raspiblitz/raspiblitz #1894: Channel Closed with Low TX Fees - Transactions Stuck in Mempool. Any way to add TX fees?

kaimeinke, 2020-12-22:
Oh my ... I made a mistake. My full node had an HDD issue and after I was able to recover the node I went to close the channels in order to prepare a migration to a new SDD (never did this before).

While I was kind of distracted I missed the proper fee settings and now my closing on-chain transactions are stuck at the bottom of the mempool ... maybe for weeks / months / years?

These are the corresponding transaction hashes of the six channels:
cd83a1f799b45fc2de34a897f864c516008c9be0133b6159b1bca57e94c54347
[image]

7e11e26a28e63fdb7bf6738ae0c995b59bb8fea47ad17f2e148bd407d3d34fda
[image]

461941d6ac765645a69f6c8fc5e163988caa34e06189a700007ec45f9c435662
[image]

ec0e9080682f9a0dd3e8e314ca3bfd5f58799c2e98b7766ac0a837b4149c4437
[image]

a8ced51fc15d48d628a6d55e1ce69ea710c0c3a2b2b901d09e5bcc21f5f00fd5
[image]

3610bd52841b792814aac2b7cb4cadd8759af46b6fd39f669e98ecf41e3bd7af
[image]

Do I have any chance to help these transactions to move on from here? Any assistance would be highly appreciated. Thank you a lot. I am running a Raspiblitz and willing to learn.

A look at the current mempool does not make me optimistic.

[image]

senf42, 2020-12-22:
The last time the mempool cleared out was nine days ago. It won't be fast, but you should not have to wait "months or years" to get your funds. I suspect things will slow down for Christmas and you'll see them get in blocks then.

kaimeinke, 2020-12-22:
Thank you a lot @senf42 , that gives me hope. I have thought there must be some clear out event as there are only <1000 transaction stuck there, but your estimate gives me confidence. Thanks and stay safe and healthy.

kaimeinke, 2020-12-27:
Hi, the mempool has been cleared after a few days and now my "close to zero transaction fee" transactions went through. Wohooo, perfect timing and this issue is closed now.

closed 2020-12-27 by kaimeinke