raspiblitz/raspiblitz #1895: Electrum Rust Server not connecting to Desktop App & appearently not syncing Index

Mirksen, 2020-12-22:
Hello,

unfortunately I cannot connect to my Electrum Rust Server from Mac and it looks like it has Sync-Problems (it was succesfully running some time before):

- tried reinstall service twice (first with new index, second kept the index)
- set ufw allow 50002
- reinstalled & restartet nginx, no success
- ping to raspberry pi works
- tried connection via 50002:s and 50001:t
- "telnet hostname 50002" succesful
- Electrum App on Client shows that Blockchainheight of my server is way lower (does that mean it connects but Index is not synced?), showing the "red connection problem bubble" on Electrum app
- 

I just reinstalled Electrum Rust Server again, still no success (this is the third time, on the second one I rebuilt Index as well):

-- Logs begin at Tue 2020-12-22 03:17:37 CET, end at Tue 2020-12-22 20:59:18 CET. --
Dec 22 20:44:48 silverbullet systemd[1]: Started Electrs.
Dec 22 20:44:48 silverbullet electrs[32535]: Config { log: StdErrLog { verbosity: Trace, quiet: false, timestamp: Millisecond, modules: [], writer: "stderr", color_choice: Auto }, network_type: bitcoin, db_path: "/mnt/hdd/app-storag
Dec 22 20:44:48 silverbullet electrs[32535]: 2020-12-22T20:44:48.949+01:00 - DEBUG - Server listening on 127.0.0.1:4224
Dec 22 20:44:48 silverbullet electrs[32535]: 2020-12-22T20:44:48.958+01:00 - DEBUG - Running accept thread
Dec 22 20:44:48 silverbullet electrs[32535]: 2020-12-22T20:44:48.960+01:00 - INFO - NetworkInfo { version: 200000, subversion: "/Satoshi:0.20.0/", relayfee: 0.00001 }
Dec 22 20:44:48 silverbullet electrs[32535]: 2020-12-22T20:44:48.961+01:00 - INFO - BlockchainInfo { chain: "main", blocks: 662536, headers: 662536, verificationprogress: 0.9999984816235892, bestblockhash: "0000000000000000000542ca9
Dec 22 20:44:48 silverbullet electrs[32535]: 2020-12-22T20:44:48.962+01:00 - DEBUG - opening DB at "/mnt/hdd/app-storage/electrs/db/mainnet"
Dec 22 20:44:49 silverbullet electrs[32535]: 2020-12-22T20:44:49.072+01:00 - TRACE - latest indexed blockhash: 00000000000000000009b3ea50248ee6f74ca44ee45f62ef661f43770972eb75
Dec 22 20:44:57 silverbullet electrs[32535]: 2020-12-22T20:44:57.779+01:00 - DEBUG - applying 662153 new headers from height 0
Dec 22 20:44:58 silverbullet electrs[32535]: 2020-12-22T20:44:58.552+01:00 - INFO - enabling auto-compactions
Dec 22 20:44:58 silverbullet electrs[32535]: 2020-12-22T20:44:58.573+01:00 - DEBUG - relayfee: 0.00001 BTC
Dec 22 20:44:58 silverbullet electrs[32535]: 2020-12-22T20:44:58.575+01:00 - DEBUG - downloading new block headers (662153 already indexed) from 0000000000000000000542ca9309e8db6a80955c9d207231a9de80ada958e386
Dec 22 20:44:58 silverbullet electrs[32535]: 2020-12-22T20:44:58.715+01:00 - TRACE - downloaded 384 block headers
Dec 22 20:44:58 silverbullet electrs[32535]: 2020-12-22T20:44:58.717+01:00 - INFO - best=0000000000000000000542ca9309e8db6a80955c9d207231a9de80ada958e386 height=662536 @ 2020-12-22T19:40:13Z (384 left to index)
Dec 22 20:46:50 silverbullet electrs[32535]: 2020-12-22T20:46:50.325+01:00 - DEBUG - applying 384 new headers from height 662153
Dec 22 20:46:50 silverbullet electrs[32535]: 2020-12-22T20:46:50.327+01:00 - WARN - reconnecting to bitcoind: disconnected from daemon while receiving
Dec 22 20:47:32 silverbullet electrs[32535]: 2020-12-22T20:47:32.323+01:00 - INFO - Electrum RPC server running on 0.0.0.0:50001 (protocol 1.4)
Dec 22 20:47:35 silverbullet electrs[32535]: 2020-12-22T20:47:35.042+01:00 - TRACE - [127.0.0.1:39214] spawned ThreadId(14)
Dec 22 20:47:35 silverbullet electrs[32535]: 2020-12-22T20:47:35.042+01:00 - INFO - [127.0.0.1:39214] connected peer
Dec 22 20:47:35 silverbullet electrs[32535]: 2020-12-22T20:47:35.043+01:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":\"electrs-check\"}\n")
Dec 22 20:47:37 silverbullet electrs[32535]: 2020-12-22T20:47:37.047+01:00 - TRACE - RPC Done
Dec 22 20:47:37 silverbullet electrs[32535]: 2020-12-22T20:47:37.047+01:00 - DEBUG - [127.0.0.1:39214] shutting down connection
Dec 22 20:47:37 silverbullet electrs[32535]: 2020-12-22T20:47:37.047+01:00 - INFO - [127.0.0.1:39214] disconnected peer
Dec 22 20:47:37 silverbullet electrs[32535]: 2020-12-22T20:47:37.331+01:00 - DEBUG - downloading new block headers (662537 already indexed) from 0000000000000000000274ef4756e0934e685ed861938ddd6598e69399bb820e
Dec 22 20:47:37 silverbullet electrs[32535]: 2020-12-22T20:47:37.332+01:00 - TRACE - downloaded 1 block headers
Dec 22 20:47:37 silverbullet electrs[32535]: 2020-12-22T20:47:37.332+01:00 - INFO - best=0000000000000000000274ef4756e0934e685ed861938ddd6598e69399bb820e height=662537 @ 2020-12-22T19:46:17Z (1 left to index)
Dec 22 20:47:37 silverbullet electrs[32535]: 2020-12-22T20:47:37.716+01:00 - DEBUG - applying 1 new headers from height 662537
Dec 22 20:47:38 silverbullet electrs[32535]: 2020-12-22T20:47:38.647+01:00 - TRACE - [77.123.105.226:37246] spawned ThreadId(17)
Dec 22 20:47:38 silverbullet electrs[32535]: 2020-12-22T20:47:38.647+01:00 - TRACE - [77.123.105.226:37246] joining ThreadId(14)
Dec 22 20:47:38 silverbullet electrs[32535]: 2020-12-22T20:47:38.647+01:00 - INFO - [77.123.105.226:37246] connected peer
Dec 22 20:47:38 silverbullet electrs[32535]: 2020-12-22T20:47:38.648+01:00 - TRACE - RPC Done
Dec 22 20:47:38 silverbullet electrs[32535]: 2020-12-22T20:47:38.648+01:00 - DEBUG - [77.123.105.226:37246] shutting down connection
Dec 22 20:47:38 silverbullet electrs[32535]: 2020-12-22T20:47:38.648+01:00 - INFO - [77.123.105.226:37246] disconnected peer
Dec 22 20:47:47 silverbullet electrs[32535]: 2020-12-22T20:47:47.815+01:00 - TRACE - [127.0.0.1:39257] spawned ThreadId(19)
Dec 22 20:47:47 silverbullet electrs[32535]: 2020-12-22T20:47:47.815+01:00 - TRACE - [127.0.0.1:39257] joining ThreadId(17)
Dec 22 20:47:47 silverbullet electrs[32535]: 2020-12-22T20:47:47.815+01:00 - INFO - [127.0.0.1:39257] connected peer
Dec 22 20:47:47 silverbullet electrs[32535]: 2020-12-22T20:47:47.815+01:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":\"electrs-check\"}\n")
Dec 22 20:47:49 silverbullet electrs[32535]: 2020-12-22T20:47:49.363+01:00 - TRACE - RPC PeriodicUpdate
Dec 22 20:47:49 silverbullet electrs[32535]: 2020-12-22T20:47:49.818+01:00 - TRACE - RPC Done
Dec 22 20:47:49 silverbullet electrs[32535]: 2020-12-22T20:47:49.818+01:00 - DEBUG - [127.0.0.1:39257] shutting down connection
Dec 22 20:47:49 silverbullet electrs[32535]: 2020-12-22T20:47:49.818+01:00 - INFO - [127.0.0.1:39257] disconnected peer
Dec 22 20:47:51 silverbullet electrs[32535]: 2020-12-22T20:47:51.548+01:00 - TRACE - [77.123.105.226:37328] spawned ThreadId(21)
Dec 22 20:47:51 silverbullet electrs[32535]: 2020-12-22T20:47:51.548+01:00 - INFO - [77.123.105.226:37328] connected peer
Dec 22 20:47:51 silverbullet electrs[32535]: 2020-12-22T20:47:51.549+01:00 - TRACE - [77.123.105.226:37328] joining ThreadId(19)
Dec 22 20:47:51 silverbullet electrs[32535]: 2020-12-22T20:47:51.551+01:00 - TRACE - RPC Done
Dec 22 20:47:51 silverbullet electrs[32535]: 2020-12-22T20:47:51.551+01:00 - DEBUG - [77.123.105.226:37328] shutting down connection
Dec 22 20:47:51 silverbullet electrs[32535]: 2020-12-22T20:47:51.551+01:00 - INFO - [77.123.105.226:37328] disconnected peer
Dec 22 20:47:55 silverbullet electrs[32535]: 2020-12-22T20:47:55.565+01:00 - TRACE - [127.0.0.1:39321] spawned ThreadId(23)
Dec 22 20:47:55 silverbullet electrs[32535]: 2020-12-22T20:47:55.565+01:00 - TRACE - [127.0.0.1:39321] joining ThreadId(21)
Dec 22 20:47:55 silverbullet electrs[32535]: 2020-12-22T20:47:55.565+01:00 - INFO - [127.0.0.1:39321] connected peer
Dec 22 20:47:55 silverbullet electrs[32535]: 2020-12-22T20:47:55.565+01:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":\"electrs-check\"}\n")
Dec 22 20:47:56 silverbullet electrs[32535]: 2020-12-22T20:47:56.593+01:00 - TRACE - [127.0.0.1:39330] spawned ThreadId(25)
Dec 22 20:47:56 silverbullet electrs[32535]: 2020-12-22T20:47:56.593+01:00 - INFO - [127.0.0.1:39330] connected peer
Dec 22 20:47:56 silverbullet electrs[32535]: 2020-12-22T20:47:56.593+01:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":\"electrs-check\"}\n")

This is the Status as of Raspiblitz>Electrum Rust Server>Status:
"
##### STATUS ELECTRS SERVICE
configured=1
serviceInstalled=1
serviceRunning=1
tipSynced=0
electrumResponding=1
initialSynced=1
localIP='192.168.178.51'
portTCP='50001'
localTCPPortActive=1
publicTCPPortAnswering=1
portHTTP='50002'
localHTTPPortActive=1
publicHTTPPortAnswering=1
TORrunning=0
nginxTest=1

Press ENTER to get back to main menu.
"
Here the tipSynced=0 seems questionable, google not helping for this one :/

Any Ideas here?

BR

leopeo13, 2021-01-09:
I had the same problem.

I reinstalled electr and kept the index, still didnt work.
I indexed again and everything works.

I do a reboot, again problems. Cannot sync. Did you restart as well?

When there is a restart is says stopping bitcoind and lighning but not electr. Maybe the electrum service is not probably stopped for a safe rebood and get corrupted?

I am now indexing again and planing not to reboot ever

rootzoll, 2021-01-12:
@leopeo13 will adding stopping electrs on reboot for v1.6.3 release - thanks for the suggestion.

rootzoll, 2021-01-12:
Pushing this to next release for review if we can improve this

bilalelreda, 2021-03-15:
Just wanted to chime in to say that I can reproduce this issue. Running RaspiBlitz v1.6.2.

openoms, 2021-03-15:
@bilalelreda the latest RaspiBlitz release is 1.6.3

You can test the PR with the latest version of electrs:
https://github.com/rootzoll/raspiblitz/pull/2059#issuecomment-799785572

rootzoll, 2021-03-16:
The PR #2059 is now merged to be part of up coming v1.7 release. Closing this issue.

closed 2021-03-16 by rootzoll