raspiblitz/raspiblitz #1896: Funding, routing and earning with this node.

maaxxyy, 2020-12-22:
Hi guys can somebody point me in the right direction as i have been trying to add peers and open channels and all the i get is errors relating to insufficient funds etc.  Like this -

not enough witness outputs to create funding transaction, need 0.0017275 BTC only have 0.0003 BTC available

not enough witness outputs to create funding transaction, need 0.00035455 BTC only have 0.0003 BTC available

<html> <head><title>504 Gateway Time-out</title></head> <body bgcolor="white"> <center><h1>504 Gateway 
Time-out</h1></center> <hr><center>nginx/1.14.2</center> </body> </html>      

So my question is how much do we need to fund the node with? As it seems like its needs a minimum of £30 worth of btc added to it.
How do we make the node into a routing node so it can earn btc on the transactions?
At the moment i have added 63000 satoshi to it so far.

Thanks

rootzoll, 2020-12-22:
"Lightning Node Management" is a complex topic ... @openoms started a GitBook with collection of resources on this: https://openoms.gitbook.io/lightning-node-management/

maaxxyy, 2020-12-23:
Thank you for that, i have some reading to do!!!

ChuckNorrison, 2020-12-31:
To respond on the topic: Earn money with lightning node?

Just running a node will not make you earn from routing fees. this needs a project behind in need to handle payments. Lightning Network is a great way for some solutions... there are a lot of projects coming up to participate on where it could be possible to serve channels to others on a kind of trading market. 

Iam running my node for around 4 months and there are around 2-3 transactions my node routed and earned around 3 sats with, even you got 15+ good connected channels. 

To get those channels you need some more money to settle on and you loose money after it got closed for reasons.

Iam running this node to solve payments for coffee via lightning. Users would use mostly LNBits instant wallets which gets filled for some fiat and be used to pay coffee with static QR code. A bit more elegant and better privacy than a public pen and paper solution in my mind.

Anyway its amazing tech and very interesting to get into it :)

nyxnor, 2021-04-18:
@maaxxyy I suppose this was solved as you did not comment anymore. Please close this issue if so.

nyxnor, 2021-05-17:
@rootzoll Please close this.

closed 2021-05-17 by rootzoll