raspiblitz/raspiblitz #1897: 1.6.2  unable to initialize tor controller

jodobear, 2020-12-22:
After a fresh reinstall of 1.6.2 and recovering LND using backup everything went well until unlocking LND wallet, restarting didn't help. i get the following error:

```sh
Dec 22 21:24:58 raspberrypi lnd[3782]: unable to initialize tor controller: unable to retrieve authentication cookie: open /run/tor/control.authcookie: permission denied
```

I've checked #1704 but, this is a permission error.

On checking #1816:

On: `curl --socks5 localhost:9050 --socks5-hostname localhost:9050 -s https://check.torproject.org/ | cat | grep -m 1 Congratulations | xargs`
Get: `Congratulations. This browser is configured to use Tor.`

So, Tor is running, but somehow there's a permission issue. I don't want to go through a reinstall if it is as trivial as changing permission for the contentious file.

Log file: https://termbin.com/sl4t

Thanks!

rootzoll, 2020-12-22:
@openoms can you take a look if that is relevant for v1.6.3 release?

openoms, 2020-12-22:
This is about the old lnd.conf not containing the Tor password.

https://t.me/raspiblitz/46276

This needs to be in the lnd.conf:

[Tor]
tor.password=passwordB

and nothing else about Tor.

Use the menu to change PasswordB to sync the passwords if not sure.

jodobear, 2020-12-22:
worked, closing issue

closed 2020-12-22 by jodobear