raspiblitz/raspiblitz #1898: 70initLND - Warning70

frisko123, 2020-12-22:
Hi Guys,

it seems like my bitcoind service can't run. I get following message: Service bitcoind is not running.

https://termbin.com/xksdr

I tried to install the sphinx-relay, maybe there was something wrong with it?
https://github.com/stakwork/sphinx-relay/blob/master/docs/raspiblitz_deployment.md

Thanks for your help in advance!

rootzoll, 2020-12-22:
can you check if you see any hint of that error with `sudo tail -n 100 /mnt/hdd/bitcoin/debug.log` ... and bitcoind worked before you installed sphinx-relay? 

BTW since RaspiBlitz v1.6.2 there is a one-click install for Sphinx-Relay under `MAINMENU > SERVICES` ... the detailed tutorial you linked is outdated. Have you tried that? Maybe best is to recover your RaspiBlitz with a fresh v1.6.2 sd card image and then use the one-click install.

frisko123, 2020-12-22:
Hey, thanks for the superfast answer. Yes, bitcoind worked before. Maybe it would indeed be the best to recover with a fresh install.

Also it seems like the raspi can't find or properly mount the ssd for some reason. There's also no directory bitcoin in hdd.
[image]

rootzoll, 2020-12-22:
Yeah seems best to give a fresh v1.6.2 sd card image a chance and use the one-click install for Sphinx-relay.

frisko123, 2020-12-23:
thanks for the help. one more thing: how can I backup funds from channels via command line? I presume the funds in my wallet will still be there after using the seed to recover the node?

rootzoll, 2020-12-23:
You can simply copy the static channel backup file to a secure place of your choosing at any time ... its located `/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup` and updates after a channel gets opened or closed.

closed 2020-12-23 by rootzoll