raspiblitz/raspiblitz #1899: Mempool add "loading blocks" info

rootzoll, 2020-12-22:
If you install on a fresh RaspiBlitz just "mempool explorer" (without any other apps) it gets stuck for some time on "loading blocks" ...

[image]

Add to info that this can happen and usually should be gone after 2hours.

NilByte, 2020-12-22:
You get the info that it could take 7 hours. This is on an installation with RTL and  a few other apps installed.

[image]

rootzoll, 2020-12-23:
There are two things ... one is waiting for the txindex to get ready and then mempool needs to gather data for some blocks until its ready - but maybe that is something for mempool to improve on their side - if "Loading blocks" takes too long it should give a some more info to the user why its taking so long.

rootzoll, 2020-12-23:
Reported on mempool github: https://github.com/mempool/mempool/issues/277
Pushing this to v1.7 release.

NilByte, 2020-12-23:
After about 16 hours I tried to deactivate and activate Mempool again. Same loading blocks message. Then activated Bitcoin Explorer and did a reboot. Mempool then loaded.

Reboot seems to fix it as a workaround.

softsimon, 2020-12-31:
We are aware of the many issues related to the now deprecated mempool v1 version. 

Please stay put for mempool v2 version for raspberry pi in the coming weeks.

closed 2021-02-21 by rootzoll