raspiblitz/raspiblitz #1900: junk characters around edges of the TERMINAL raspiblitz menu display

agr999, 2020-12-23:
Menu command are fine and can select/unselect but there exist "junk" characters (i.e. were not there originally) around the edges of the display. Problem appeared after aborting an Electrs server sync.

RaspiBlitz functionality does not appear to be affected. It's only bothersome to look at. ;) Just wondering where this junk comes from and perhaps how to get rid of it.

v1.6.2

NilByte, 2020-12-23:
If you use Putty, try this.

[image]

nyxnor, 2021-04-18:
@agr999 Please close this issue if solved.

closed 2021-04-18 by agr999

chGoodchild, 2023-06-27:
Perhaps this has something to do with how the encoding is interpreted on the terminal that ssh's into the machine. A [similar problem](https://nostr.build/i/8e7347cd1f07b4ca59469719d3fd48e01ee251d24db7da7b80659bd4fe4e233a.jpg) was discovered when logging into a raspiblitz from the terminal of a chromebook. The problem didn't occur logging in to the same pi from termux on an android, though.

rootzoll, 2023-07-29:
@chGoodchild yes this problem occurs based on the SSH client/terminal people use.