raspiblitz/raspiblitz #1901: Bitcoind: Can't open PID from file

jodobear, 2020-12-24:
Fresh 1.6.2 install with LND recovered from backup. Bitcoind is `active` but it isn't syncing, get `Failed to parse PID from file` and after service restart get `Can't open PID from file` as shown in the below `status` output:
```sh
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-12-23 12:20:38 GMT; 1min 32s ago
  Process: 10999 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/
  Process: 11001 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/
 Main PID: 11002 (bitcoind)
    Tasks: 18 (limit: 4915)
   Memory: 199.2M
   CGroup: /system.slice/bitcoind.service
           └─11002 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin

Dec 23 12:20:38 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 23 12:20:38 raspberrypi systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?)
Dec 23 12:20:38 raspberrypi systemd[1]: Started Bitcoin daemon.
```

```sh
$ ls -l /home/bitcoin/.bitcoin/
total 324
-rw------- 1 bitcoin bitcoin     37 Dec 22 20:22 banlist.dat
-r--r--r-- 1 root    root       854 Dec 22 23:56 bitcoin.conf
-rw------- 1 bitcoin bitcoin      5 Dec 24 08:43 bitcoind.pid
drwx------ 1 bitcoin bitcoin 114010 Dec 21 16:33 blocks
drwx------ 1 bitcoin bitcoin  48572 Dec 24 08:43 chainstate
-rw------- 1 bitcoin bitcoin   6877 Dec 24 08:49 debug.log
-rw------- 1 bitcoin bitcoin 247985 Dec 23 15:30 fee_estimates.dat
-rw------- 1 bitcoin bitcoin     17 Dec 23 15:30 mempool.dat
-rw------- 1 bitcoin bitcoin    820 Dec 24 08:43 onion_private_key
-rw------- 1 bitcoin bitcoin  49552 Dec 23 15:30 peers.dat
```

rootzoll, 2020-12-24:
PR #1902 merged .. this should delete the blocking file on boot. Will be part of v1.6.3 release - closing issue.

jodobear, 2020-12-24:
so, can i just safely delete the `pid & `lock` file and start bitcoind to resolve the issue?

rootzoll, 2020-12-24:
@jodobear delete those files and do `restart` should be fine.

jodobear, 2020-12-24:
> @jodobear delete those files and do `restart` should be fine.

That didn't work.

syslog: `Dec 24 22:43:17 raspberrypi systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory`

status:
```
Dec 24 22:43:17 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 24 22:43:17 raspberrypi systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?)
Dec 24 22:43:17 raspberrypi systemd[1]: Started Bitcoin daemon.
```

`ls -l /home/bitcoin/.bitcoin/`: `-rw------- 1 bitcoin bitcoin      5 Dec 24 22:43 bitcoind.pid`

rootzoll, 2020-12-25:
Maybe try stopping bitcoind first and then restart:
```
sudo systemctl stop bitcoind
sudo rm -f /home/bitcoin/.bitcoin/bitcoind.pid
restart
```

jodobear, 2020-12-25:
That's exactly what i did, but didn't help

rootzoll, 2020-12-25:
Hmm .. if the file is deleted and really gone before restart then it gets created on start again. Bitcoind systemd service is creating that PID file ... so does it gets started twice? can you check with `sudo systemctl status bitcoind` and `htop` if there is a bitcoind running already. Also check the process id stored in the PID file and check if that proces sis still running with `sudo ps $(sudo cat /home/bitcoin/.bitcoin/bitcoind.pid)`

jodobear, 2020-12-26:
After stopping bitcoin and removing pid file, there is no bitcoin running. On restart the issue persists. On checking processes, there's only 1 bitcoind instance:

[image]

jodobear, 2020-12-26:
I flashed a fresh 1.6.2 microSD. The blitz configured it, reboot 2-3 times, all seemed well but, **the issue still persists**

jodobear, 2020-12-26:
I flashed a new 1.6.2 card. The blitz confgured itself, rebooted a couple of times, set my passwords and everything seemed fine but, **the issue persists**. 

There is no syslog for bitcoind on restarting blitz after deleting pid file. But, simply restarting bitcoind syslog gives:
```sh
Dec 26 11:24:33 raspberrypi systemd[1]: bitcoind.service: Failed to parse PID from file /home/bitcoin/.bitcoin/bitcoind.pid: Invalid argument
```

jodobear, 2021-01-05:
Thank you for re-opening the issue. My node is still down. Not sure what to do anymore. Let me know how can i help resolve this.

NilByte, 2021-01-09:
Maybe check if wait for network connection is set in `sudo raspi-config` ?

S6 Network at Boot   Select wait for network connection on boot

It should be by default on the raspiblitz configuration.

rootzoll, 2021-01-10:
> Thank you for re-opening the issue. My node is still down. Not sure what to do anymore. Let me know how can i help resolve this.

@jodobear I would give it a try to remove the PID file at all. Go to terminal:
`sudo systemctl disable bitcoind` and then edit the systemd servcie for example with `sudo nano /etc/systemd/system/bitcoind.service` remove the `-pid=...` parameter from ExecStart and save edit (CRTL+o) and exit (CTRL+x). Then `sudo systemctl enable bitcoind` and `restart` system.

Let me know if this fixes the problem for you.

rootzoll, 2021-01-10:
@openoms do you see any downsides to remove the PID file from the system service?

openoms, 2021-01-10:
@jodobear I would want make sure that odd problem is not caused by a failing SDcard.

Could you try flashing the v1.6.2 image on a new SDcard?

rootzoll, 2021-01-12:
We are keeping this issue open because it has a pending PR #1930 .. but moving it to next release because the fix would be to cloase to final realease of v1.6.3 to include.

NilByte, 2021-02-01:
After updating to 1.6.3 i got this error in debug.

Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory

```
[H[2J[3J
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.3
chainnetwork: bitcoin / main
 17:46:32 up 24 min,  2 users,  load average: 2.61, 2.14, 1.87

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-02-01 17:25:19 GMT; 21min ago
  Process: 3887 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 3892 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 3959 (bitcoind)
    Tasks: 19 (limit: 4915)
   Memory: 5.0G
   CGroup: /system.slice/bitcoind.service
           └─3959 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Feb 01 17:25:19 raspberrypi systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory
Feb 01 17:25:19 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2021-02-01 17:21:40 GMT, end at Mon 2021-02-01 17:46:32 GMT. --
Feb 01 17:25:19 raspberrypi systemd[1]: Starting Bitcoin daemon...
Feb 01 17:25:19 raspberrypi systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory
Feb 01 17:25:19 raspberrypi systemd[1]: Started Bitcoin daemon.

```

Set wait for network connection in `sudo raspi-config`

Rebooted and ran debug again and got rid of the Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory

```
[H[2J[3J
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.3
chainnetwork: bitcoin / main
 19:52:35 up 7 min,  2 users,  load average: 3.65, 2.77, 1.33

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-02-01 19:47:55 CET; 4min 40s ago
  Process: 6572 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 6579 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 6728 (bitcoind)
    Tasks: 19 (limit: 4915)
   Memory: 3.8G
   CGroup: /system.slice/bitcoind.service
           └─6728 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Feb 01 19:47:54 raspberrypi systemd[1]: Starting Bitcoin daemon...
Feb 01 19:47:55 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2021-02-01 19:44:29 CET, end at Mon 2021-02-01 19:52:35 CET. --
Feb 01 19:47:54 raspberrypi systemd[1]: Starting Bitcoin daemon...
Feb 01 19:47:55 raspberrypi systemd[1]: Started Bitcoin daemon.
```

I don't know if setting the wait for network on boot is what fixed it, but it might be worth a try.

rootzoll, 2021-03-23:
The #1930 was also not a solution ... pushing this to v1.7.1 for another fresh look.

rootzoll, 2021-04-17:
I had a test machine with that problem again.
```
Apr 17 19:12:51 raspberrypi systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory
```

For v1.7 I  give it another try with changing the path of the PID file patch from `/home/bitcoin/.bitcoin` to `/mnt/hdd/bitcoin` to eliminate that this has something to do with this directory being "linked".

If that problem shows up again, I would recommend to use `/home/bitcoin` so that its neither a linked directory nor on the HDD.

netsuke1, 2021-05-12:
since yesterday RaspiBlitz ... unfortunately the bitcoind service is not running

https://termbin.com/jl7w

openoms, 2021-05-12:
> since yesterday RaspiBlitz ... unfortunately the bitcoind service is not running
> 
> https://termbin.com/jl7w

@netsuke1 
`Error: Error reading configuration file: parse error on line 1: bitcoind configuration`
your `/mnt/hdd/bitcoin/bitcoin.conf` is corrupted.
edit with:
`sudo nano /mnt/hdd/bitcoin/bitcoin.conf`

Likely a hardware error.
See you recovery options in the FAQ.

netsuke1, 2021-05-12:
@openoms
Thanks, how should a clean bitcoin.conf look like? 
/mnt/hdd/bitcoin/bitcoin.conf and
/home/bitcoin/.bitcoin/bitcoin.conf are identical
I can't find anything in the FAQ

[image]

openoms, 2021-05-12:
@netsuke1 delete the first line and reboot

netsuke1, 2021-05-12:
@openoms 
Thank you, you are great ... Node is syncing... actually I had ordered a pre-synced RaspiBlitz ... but what the heck

[image]

openoms, 2021-06-29:
We are using:
`KillMode=process`
in the systemd service files which is not recommended and can cause leftover subrocesses likely causing this issue.

https://www.man7.org/linux/man-pages/man5/systemd.kill.5.html

Will remove all the `KillMode=` entries so it defaults to `control-group`.

rootzoll, 2021-10-09:
Worked correct during release candidates ... closing for final v1.7.1 release.

closed 2021-10-09 by rootzoll