raspiblitz/raspiblitz #1906: Test 1.6.3 Release Candidate

rootzoll, 2020-12-25:
Download Release Candidate on your test/development node:

https://raspiblitz.org/raspiblitz-v1.6.3rc2-2020-12-24.img.gz
SHA: c11816fd8ffda991991bf687ce1fd70ceeb6c61fb960e633b1bd5a885501d0e0

Please give feedback. Thx.

Cranky4u, 2020-12-28:
Having a crack to see if resolves i2c hanging issue on boot for RPi 3b+

Cranky4u, 2020-12-28:
> 
> 
> Having a crack to see if resolves i2c hanging issue on boot for RPi 3b+

1st attempt with 1.6.3 unsuccessful - RPi will not boot
* with or without HDMI
* imaged with balenaEtcher
* system hangs on red boot light

Now making image with RUFUS and testing boot up.

Cranky4u, 2020-12-29:
> 
> 
> > Having a crack to see if resolves i2c hanging issue on boot for RPi 3b+
> 
> 1st attempt with 1.6.3 unsuccessful - RPi will not boot
> 
>     * with or without HDMI
> 
>     * imaged with balenaEtcher
> 
>     * system hangs on red boot light
> 
> 
> Now making image with RUFUS and testing boot up.
RUFUS built SD card fails too :(

My RPi 3b+ still not running with 1.6.3 or 1.6.2 or 1.6 or 1.5

Not sure what is wrong....Pi does work though as use it for retropi gaming

Dikay900, 2020-12-29:
things tested after update from v1.6.2

- headless (no monitor)
- bitcoin + lnd all over tor
- joinmarket (needed to reconfigure maker configuration / fees)
- rtl
- lndmanage
- pool

Cranky4u, 2020-12-29:
Looks like 1.6.3 is not working on my RPi 3 & Seagate 1TB drive. Hangs on HDD detection and has been left to run overnight for ~12 hours. With HDD removed, system hangs on i2c driver stage...
Maybe the issue is with compatibility with new Seagate drive?
@rootzoll Should the Seagate 1TB drive be compatible out of the box or should it be formatted first (use Win7 PC to set it up or does raspiblitz handle reformatting / set up)?

Cranky4u, 2020-12-29:
> 
> 
> working without a problem, things tested after update from v1.6.2 on rpi 4
> 
>     * headless (no monitor)
> 
>     * bitcoin + lnd all over tor
> 
>     * joinmarket (needed to reconfigure maker configuration / fees)
> 
>     * rtl
> 
>     * lndmanage
> 
>     * pool

What type of HDD do you have on your RPi 4? My RPi 3b looks like having issues with Seagate 1TB...

Dikay900, 2020-12-29:
@Cranky4u you can edit your posts rather than post multiple answers one after another just FYI

We would need the logs you are seeing for us to help you. I am using a 2TB wd external hdd drive but i think you rather have a problem with your hardware.

Also if you are being stuck on hdd detection but are able to login into the terminal please run ./XXdebugLogs.sh

Cranky4u, 2020-12-30:
> 
> 
> @Cranky4u you can edit your posts rather than post multiple answers one after another just FYI
> 
> We would need the logs you are seeing for us to help you. (Whatever you are seeing that you know you are stuck at the hdd detection) I am using a 2TB wd external hdd drive but i think you rather have a problem with your hardware.
> 
> Also if you are being stuck on hdd detection but are able to login into the terminal please run ./XXdebugLogs.sh and post the output here.

@Dikay900 Thnx for the edit posts....new to github commenting and will adjust to reflect this feedback.

@Dikay900 @rootzoll 
I am unable to digitally grab logs as it hangs and not sure how to grab them without ssh...new to linux game but have way back to winxp / dos for PC so expect to get my head around it quick enough.
This test session I reformatted HDD to exFat but still no improvement on hanging as per screen photo.
Of note, when using raspian, the HDD is mountable.

So what do you guys think i sthe next step or should I throw it in and just go for a RPi 4?

@rootzoll what RPi are you developing on? To avoid going in circles, I will probably grab new hardware based on your developer config...

[image]

openoms, 2021-01-01:
@Cranky4u try booting without anything attached to the HDMI and use ssh then.

Cranky4u, 2021-01-02:
> 
> 
> @Cranky4u try booting without anything attached to the HDMI and use ssh then.

@openoms tried no HDMI and just LAN then SSH....failed :( see more > https://github.com/rootzoll/raspiblitz/issues/1906#issuecomment-752306505

openoms, 2021-01-02:
@Cranky4u this is likely a hardware problem. Does the setup work with a 1.6.2 image?
Your HDD is buily for USB3, the RPI3 probably cannot power it properly.
Use a powered USB hub for the RPI3, but tbh switching to RPi4 would make more sense. The Rpi3 is insufficient in many ways.

frennkie, 2021-01-06:
The apt repository for Tor has an issue with the key:

```
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://deb.torproject.org/torproject.org buster InRelease: The following signatures were invalid: EXPKEYSIG 74A941BA219EC810 deb.torproject.org archive signing key
W: Failed to fetch https://deb.torproject.org/torproject.org/dists/buster/InRelease  The following signatures were invalid: EXPKEYSIG 74A941BA219EC810 deb.torproject.org archive signing key
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

I think we anyway discovered, that they no longer provide packages for armhf - right? So this should not be included anyway..?

openoms, 2021-01-06:
@frennkie 
Solved from  here:
https://support.torproject.org/apt/tor-deb-repo/
```
wget -qO- https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --import
gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | sudo apt-key add -
```

This error:
```
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://deb.torproject.org/torproject.org buster InRelease' doesn't support architecture 'armhf'
```

 is solved by changing the repos to:
```
#deb https://deb.torproject.org/torproject.org buster main
#deb-src https://deb.torproject.org/torproject.org buster main
deb [arch=amd64] https://deb.torproject.org/torproject.org buster main
deb-src [arch=amd64] https://deb.torproject.org/torproject.org buster main
```
Will make a PR.

FreePietje, 2021-01-09:
>is solved by changing the repos to:
>

> #deb https://deb.torproject.org/torproject.org buster main
> #deb-src https://deb.torproject.org/torproject.org buster main
> deb [arch=amd64] https://deb.torproject.org/torproject.org buster main
> deb-src [arch=amd64] https://deb.torproject.org/torproject.org buster main

Yeah, you REALLY shouldn't do that.  
You can't run packages made for AMD64 on an armhf system. Some packages are `Architecture: all` and for those it doesn't matter, but for all the rest, you need packages made for your architecture (armhf).
Tor's APT repo does also support arm64, so if you've switched raspiblitz to use arm64 (which both RPi3B and 4 support), **then** you can use their APT repo.

openoms, 2021-01-09:
@FreePietje 
testing so far showed no problems and this is the recommendation (at least for Focal)
from the torproject website:
```
Note: Ubuntu Focal dropped support for 32-bit, so instead use:

deb [arch=amd64] https://deb.torproject.org/torproject.org focal main deb-src [arch=amd64] https://deb.torproject
```
Do you think we rather should use the fallback version which is a major version behind?

In any case the v1.6.3 is planned to be the last release based on 32 bit and we have an option to build Tor from source in the UPDATE menu.

FreePietje, 2021-01-09:
The instructions you're referring to are that Focal (apparently) doesn't support 32-bit `i386` architecture anymore. As both Intel and AMD CPUs from (at least) the last 10 years (also) support the 64-bit `amd64` architecture, that'll probably work.
There's a (very) good chance that Focal does support the 32-bit `armhf` architecture (and 64-bit `arm64`).  
But i386/amd64 and armhf/arm64 are very different architectures.

https://wiki.debian.org/Multiarch/HOWTO explains how it is supposed to work (arch qualifiers for `deb-src` line don't make any sense btw)

My guess is that it actually still doesn't work, only that you haven't triggered the problem (yet). You can put all kinds of 'crap' in your sources.list, but if you don't do an `apt/aptitude update`, that all doesn't matter.
For `muti-arch` to work, you need to do `dpkg add-architecture <arch>` and I haven't found that in this project.
It could also be that `apt` still won't complain, but doing `apt install tor:amd64` really should fail.

> Do you think we rather should use the fallback version which is a major version behind?

I don't know which version you're using now, but the usual way to get newer versions of a package for a stable release, is using 'backports'. Not all packages have backports, but tor does: https://packages.debian.org/buster-backports/tor and through that you can install the `armhf` version for RPi 2/3/4. For RPi 0/1 you could use armel, but that's not optimal (for RPi's)

openoms, 2021-01-09:
@FreePietje thank you for the detailed explanation, it make sense. I have actually mistaken the `amd64` for `arm64`, but still it is not a good way to deal with this. Surprised it worked likely because it had no effect.
Removed the referene to `amd64` in https://github.com/rootzoll/raspiblitz/pull/1919/commits/6bb1dba0bf77c10d38636149975b130ea73e8aff

Ben10Brasileiro, 2021-01-10:
> Openoms to Cranky4u
> @Cranky4u try booting without anything attached to the HDMI and use ssh then.

I think I found a solution for `HDMI `and `I2C /dev entries driver` (the second one i2c is already in rapi-config menu, so not sure if it is failing for people that have chosen this setup, see more below).
Im still downloading the 1.6.3.... sorry if it´s not apropriate to do this on this thread.
Decided to download the 1.6.3, in the next days I will test it for having more accuracy on my comments, will edit if something is not compatible on 1.6.3 with what Im saying.
As I dont have an  LCD, I cannot verify that, correct me if im wrong with something, but thats what worked for me below.

---

My setup that works on boot (didnt need to disconnect anything or press ctrl+D):
HDMI plugged
FAN plugged (5V pins)
SSD with USB ADAPTER plugged (UBS 3.0) - both recommended in README
USB keyboard plugged
LAN plugged
MICRO SD plugged (boots from microSD)

---
HDMI >

Problem and research: not detecting hdmi, doesnt require the user to unplug and plug the cable or create a HDMI file.

What I tried and didnt work:
HDMI file
HDMI (command)
sudo /home/admin/config.scripts/blitz.lcd.sh hdmi on (same as above)

Solution (with this, there is NO need to unplug hdmi on boot): 
in `config.txt` should be like this (it enables a lot of requirements automatically, see https://www.raspberrypi.org/documentation/configuration/config-txt/video.md)
#uncomment if you get no picture on HDMI for a default "safe" mode
`hdmi_safe=1` (uncomment this line, it enables 10 more things for HDMI)

Comments: I had to unplug at boot and plug again before hdmi_safe=1, now I can maintan the cable attached during boot and the hdmi works. I tried creating hdmi file (didnt work), tried the command HDMI (didnt work either). The only solution that worked (besides unplug at boot, which needs to be done manually), the `hdmi_safe=1` at config.txt is the best option. I dont have a pi LCD screen, but even with the HDMI I was having these problems (I didnt start even after 30 minutes, had to turn the power off several times, even flashed 3 times), maybe because I2C refers to the pins of the fan.

---
I2C /dev entries driver >

Problem and reserch: hangs on boot, either doesnt start even with Alt+F2(problem 1) or just start after Alt+F2 or Ctrl+D (problem 2: sometimes Ctrl+D doesnt appear and even if it doesnt appear and I press Ctrl+D doesnt work somestimes, but sometime can bypass with Ctrl+D, but it´s manually done). Searched a lot of forums, official site, each one gives differents solutions

Solution:

Solution 1 (this worked for me, also tried several boots, unplugging and plugging HDMI, Ctrl+D appeared)
`sudo raspi-config`
5 Interfacing Options
P5 I2C
'Yes' (enable automatic loading of I2C on boot) (this does insert i2c-dev to modules and 'dtparam=i2c_arm=on', which is already in your default config file, but why not insist)

Solutions 2 (I tried to revert solution 1 so the i2c error would appear, so I could test the following, but now it just stay 5secs at the i2c /dev entries driver message and boots without needing to Ctrl+D. So, the following code, I didnt test)
on `config.txt`, continue with `dtparam=i2c=on` and add (I now some are default by the raspi, but, in hdmi some were default and just worked if it was mentioned on code, so Im not relying on default that is not written on file) (why persist if its already default? Its default on rpi4, but not all things are default on rpi3):
`ignore_lcd=0` (makes sure to detect LCD)
`display_default_lcd=1` (makes LCD the default display)
`disable_overscan=0`
`max_framebuffers=2`
`framebuffer_priority=0 (perfer main lcd)`

rootzoll, 2021-01-12:
@Ben10Brasileiro Thanks for the deep research into the i2c /den entries problem - thats a great lead to look deeper into. Basically a lot of those settings could be part of the sd card image build. I created a new ticket on this #1940 to push it one release up, because its a bit to short notice to put that into the v1.6.3 final release.

rootzoll, 2021-01-13:
@openoms @frennkie or anybody with time and a testnode on your hands ... I am giving this sd card image build a last etst ride, any further test installs welcome:
https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-12.img.gz
https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-12.img.gz.sig
SHA-256: 94b62b4b424c7408bd98e2ad2d118e084f9334ea5e542559248f2d61cd672f89

Ben10Brasileiro, 2021-01-13:
@rootzoll Running this latest version right now, no problems.

rootzoll, 2021-01-13:
OK I think I found a last small but stopping problem and will do another build for v1.6.3 release tomorrow.
https://github.com/rootzoll/raspiblitz/issues/1942

Cranky4u, 2021-01-13:
> 
> 
> Download Release Candidate on your test/development node:
> 
> https://raspiblitz.org/raspiblitz-v1.6.3rc2-2020-12-24.img.gz
> SHA: c11816fd8ffda991991bf687ce1fd70ceeb6c61fb960e633b1bd5a885501d0e0
> 
> Please give feedback. Thx.

Tested but my HDD is still not compatible. HDD works on Raspian but off to get a powered USB hub tomorrow so can rule that out soon. 

Of note - booting and SSH was very quick :+1: 
[image]

Cranky4u, 2021-01-13:
> 
> 
> @openoms @frennkie or anybody with time and a testnode on your hands ... I am giving this sd card image build a last test ride, any further test installs welcome:
> https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-12.img.gz
> https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-12.img.gz.sig
> SHA-256: 94b62b4b424c7408bd98e2ad2d118e084f9334ea5e542559248f2d61cd672f89

Tested but my HDD is still not compatible. HDD works on Raspian but off to get a powered USB hub tomorrow so can rule that out soon.

Of note - booting and SSH was very quick +1
[image]

openoms, 2021-01-13:
@Cranky4u can you share your power supply, USB adapter and HDD type and if you have info about it's peak power requirement?
It should not affect the recognition of the disk, but a formatting might worth to try  (with default settings on windows, NTFS is ok, just make sure there is only one big partition).

rootzoll, 2021-01-13:
@openoms @frennkie ok next v1.6.3 build is up ... giving it last testing:
https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-13.img.gz
https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-13.img.gz.sig
SHA-256: 94b62b4b424c7408bd98e2ad2d118e084f9334ea5e542559248f2d61cd672f89

openoms, 2021-01-13:
@rootzoll first the PGP dignature is ok, but I get a different sha256sum, can you double check?
```
$ gpg --verify raspiblitz-v1.6.3-2021-01-13.img.gz.sig 
gpg: assuming signed data in 'raspiblitz-v1.6.3-2021-01-13.img.gz'
gpg: Signature made Wed 13 Jan 2021 16:16:08 GMT
gpg:                using RSA key C0EE614531A416B4CDB7A2D7AA9DD1B5CC5647DA
gpg: Good signature from "Christian Rotzoll <christian@rotzoll.de>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 92A7 46AE 33A3 C186 D014  BF5C 1C73 060C 7C17 6461
     Subkey fingerprint: C0EE 6145 31A4 16B4 CDB7  A2D7 AA9D D1B5 CC56 47DA
     
$ sha256sum raspiblitz-v1.6.3-2021-01-13.img.gz
4f12f95465b4305ced8ed50feb540e0b88ff20f19f59347c8bee8712f00959c1  raspiblitz-v1.6.3-2021-01-13.img.gz

openoms, 2021-01-13:
Recovering all services caught an edge case -> #1945

rootzoll, 2021-01-13:
> the PGP signature is ok, but I get a different sha256sum, can you double check?

@openoms you are correct - that was a copy past error from the previuos image build. I now corrected in the comment above the checksum.

About the #1945 ... because its an edge case I would push it to the next release. Would like to get the release out for now ... already were a series of RC and rebuilds.

openoms, 2021-01-13:
@rootzoll other then that, all services are set up ok now!

rootzoll, 2021-01-14:
OK RaspiBlitz v1.6.3 is out :D https://twitter.com/rootzoll/status/1349552661838897153?s=20

closed 2021-01-14 by rootzoll