raspiblitz/raspiblitz #1907: JoinMarket: ModuleNotFoundError: No module named 'PySide2'

petrhejna, 2020-12-26:
I have similar issue as described here: https://github.com/rootzoll/raspiblitz/issues/1856

I got error
```
joinmarket@*******************'s password: 
Traceback (most recent call last):
  File "joinmarket-clientserver/scripts/joinmarket-qt.py", line 27, in <module>
    from PySide2 import QtCore
ModuleNotFoundError: No module named 'PySide2'
```

- I have installed JoinMarket via menu.
- I am using ssh command provided by `jm` command on Raspi
- I use it on `Ubuntu 20.04.1 LTS`
- I was not sure if the module is missing from Raspi, or on my machine so I did:
```
sudo apt-get install python3-pip
pip3 install PySide2

# And also followed this: https://github.com/JoinMarket-Org/joinmarket-clientserver/issues/763#issuecomment-742466929
# and "fixed" my python
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 1
sudo update-alternatives --install /usr/bin/pip pip /usr/bin/pip3 1
```

But no luck so far.

openoms, 2020-12-28:
Will look at this in a couple of days in.more detail.

In the meantime you can try to reinstall JoinMarket from either the raspiblitz or joininbox menu.

The Pyside2 package is installed differently for ARM and not tested with Ubuntu.

You could also try the official joinmarket installation method also.
The QT is not even necessary to use the CLI.

mmarsalko, 2021-02-08:
There's definitely some Bad Stuff™️ going on when installing.. looks like some packages are missing.

```
dialogcancel(0)
Loop Setting unchanged.
RTL Webinterface Setting unchanged.
BTC-RPC-Explorer Setting unchanged.
Cryptoadvance Specter Setting unchanged.
ElectRS Setting unchanged.
BTCPayServer setting not changed.
lndmanage setting unchanged.
faraday setting unchanged.
Balance of Satoshis setting unchanged.
PyBLOCK setting unchanged.
ThunderHub setting unchanged.
LNbits setting unchanged.
JoinMarket setting changed ..
*** INSTALL JOINMARKET ***
# OK, running behind Tor
Multiwallet is active and wallet.dat is used.
The bitcoin core wallet is already on
# cleaning before install
# add the 'joinmarket' user
Adding user `joinmarket' ...
Adding new group `joinmarket' (1008) ...
Adding new user `joinmarket' (1008) with group `joinmarket' ...
Creating home directory `/home/joinmarket' ...
Copying files from `/etc/skel' ...
# setting PASSWORD_B as the password for the 'joinmarket' user
Adding user `joinmarket' to group `sudo' ...
Adding user joinmarket to group sudo
Done.
joinmarket ALL=(ALL) NOPASSWD:ALL
# specified to use wallet.dat in the recovered joinmarket.cfg
# installing ARM specific dependencies to run the QT GUI
Reading package lists... Done
Building dependency tree       
Reading state information... Done
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1).
zlib1g-dev set to manually installed.
The following package was automatically installed and is no longer required:
  libbluetooth3
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  libjpeg62-turbo-dev libpyside2-py3-5.11 libqt5designer5 libqt5help5 libqt5qml5 libqt5test5 libshiboken2-py3-5.11 python3-sip
Suggested packages:
  qt5-qmltooling-plugins python3-pyqt5-dbg python-pyside2-doc
The following NEW packages will be installed:
  libjpeg-dev libjpeg62-turbo-dev libpyside2-py3-5.11 libqt5designer5 libqt5help5 libqt5qml5 libqt5test5 libshiboken2-py3-5.11 python3-pyqt5 python3-pyside2.qtcore python3-pyside2.qtgui
  python3-pyside2.qtwidgets python3-sip
0 upgraded, 13 newly installed, 0 to remove and 128 not upgraded.
Need to get 236 kB/8,641 kB of archives.
After this operation, 40.2 MB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libjpeg62-turbo-dev armhf 1:1.5.2-2+b1
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf libjpeg-dev all 1:1.5.2-2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libj/libjpeg-turbo/libjpeg62-turbo-dev_1.5.2-2+b1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libj/libjpeg-turbo/libjpeg-dev_1.5.2-2_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
# installing JoinMarket
Cloning into 'joinmarket-clientserver'...
remote: Enumerating objects: 195, done.
remote: Counting objects: 100% (195/195), done.
remote: Compressing objects: 100% (124/124), done.
remote: Total 8303 (delta 95), reused 136 (delta 71), pack-reused 8108
Receiving objects: 100% (8303/8303), 6.84 MiB | 5.08 MiB/s, done.
Resolving deltas: 100% (5508/5508), done.
HEAD is now at a087d26 Merge #701: add release notes and update version 0.7.1
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /home/joinmarket/joinmarket-clientserver/jmvenv/bin/python3
Also creating executable in /home/joinmarket/joinmarket-clientserver/jmvenv/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'PySide2'
X11 connection rejected because of wrong authentication.
X11 connection rejected because of wrong authentication.
X11 connection rejected because of wrong authentication.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in ./jmvenv/lib/python3.7/site-packages (21.0.1)
X11 connection rejected because of wrong authentication.
X11 connection rejected because of wrong authentication.
X11 connection rejected because of wrong authentication.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in ./jmvenv/lib/python3.7/site-packages (53.0.0)
~/joinmarket-clientserver/deps ~/joinmarket-clientserver
~/joinmarket-clientserver/deps/cache ~/joinmarket-clientserver/deps ~/joinmarket-clientserver
sha256sum: 0d9540b13ffcd7cd44cc361b8744b93d88aa76ba.tar.gz: No such file or directory
0d9540b13ffcd7cd44cc361b8744b93d88aa76ba.tar.gz: FAILED open or read
sha256sum: WARNING: 1 listed file could not be read
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   164  100   164    0     0    788      0 --:--:-- --:--:-- --:--:--   792
100  168k    0  168k    0     0   288k      0 --:--:-- --:--:-- --:--:-- 2292k
0d9540b13ffcd7cd44cc361b8744b93d88aa76ba.tar.gz: OK
~/joinmarket-clientserver/deps ~/joinmarket-clientserver
~/joinmarket-clientserver/deps/secp256k1-0d9540b13ffcd7cd44cc361b8744b93d88aa76ba ~/joinmarket-clientserver/deps ~/joinmarket-clientserver
make: *** No rule to make target 'clean'.  Stop.

```

openoms, 2021-02-08:
Just testing on a RaspiBlitz v1.6.3 (32bit) and cannot reproduce the errors.

Will test from a clean install.

@petrhejna the PySide2 package is imported form the system-site-packages for Raspbian (Debina) I would need to build an Ubuntu image on the RPi to test this, but instead we are planning to move to the 64bit Raspbian. Would recommend to base your image on that: https://github.com/rootzoll/raspiblitz/issues/1199#issuecomment-736564381

openoms, 2021-02-08:
CAn confirm the bug on the 32bit Raspbian fresh install.

Tried: 
`sudo apt install --fix-missing  libjpeg-dev libjpeg62-turbo-dev libpyside2-py3-5.11 libqt5designer5 libqt5help5 libqt5qml5 libqt5test5 libshiboken2-py3-5.11 python3-pyqt5 python3-pyside2.qtcore python3-pyside2.qtgui`

But it comes down to these packages missing from the raspbian repos:
```
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libjpeg62-turbo-dev armhf 1:1.5.2-2+b1
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf libjpeg-dev all 1:1.5.2-2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libj/libjpeg-turbo/libjpeg62-turbo-dev_1.5.2-2+b1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libj/libjpeg-turbo/libjpeg-dev_1.5.2-2_all.deb  404  Not Found [IP: 93.93.128.193 80]
```

Will test on a new 64bit install as well (there is no error my running nodes), but likely this issue will be fixed by moving to 64bit in the next major RaspiBlitz release.

There is a 64bit image which can be tested already: https://github.com/rootzoll/raspiblitz/issues/1199#issuecomment-737305510

openoms, 2021-02-11:
This is not a problem on 64 images.
The 64 bit RaspberryOS and the pue debian botu work well.
Will keep the issue open until the new release, but not looking to fix for 32bit.
(if someone finds a solution it would be welcome of course)

closed 2021-07-21 by openoms