raspiblitz/raspiblitz #1909: Lots of DNS queries on 1.6.2

Houtek, 2020-12-27:
Reference: #1656

Hi, 
even with the latest 1.6.2 release I see a lot of DNS requests on my pihole coming from Raspiblitz.

[image]

I disabled the IPv6 address on the eth0 interface to make sure, that v6 is not the root cause.

Do you need more information? I'm glad, when I can help.

Houtek, 2020-12-31:
After I checked /etc/hosts and made sure, the hostname is set correctly, the queries went to normal. 
I'm not sure, which name is set during install, but this might be worth a view.

rootzoll, 2021-01-04:
@Houtek what was the correct way to set `/etc/hosts` ?

Houtek, 2021-01-09:
Ok, forget what I said earlier. It is still very chatty.

[image]

The only way I could get rid of the amount of queries was to install a local unbound as resolver. Since I don't have a spare RPi at the moment, I can't test the behavior with a brand new install.

rootzoll, 2021-01-10:
Please update your findings. Something to keep eye on - pushing this to a later realease.

rschueler, 2021-02-07:
I can see the same effect on my pihole. The in-addr.arpa PTR request are DNS reverse lookups and I identified the IPs to be peers connected to the bitcoin node. The domain names show the IPs in reverse order: 70.229.234.44.in-addr.arpa will show up as 44.234.229.70 in Bitcoin Explorer.
My guess is, that a process related to Bitcoin Core or Bitcoin Explorer tries to lookup the domain name of connected peers for logging purpose.

PatrickScheich, 2021-02-13:
the origin of these reverse DNS  requests is a call to `netstat -a` in file `\home.admin\config.scripts\bonus.electrs.sh` as this call will try to resolve all IP addresses found. 

Prevent this by using `netstat -an`

rootzoll, 2021-03-23:
The netstat PR is merged and will be part of v1.7RC1 for final testing.

rootzoll, 2021-04-16:
@PatrickScheich @Houtek any results on this with testing v1.7RC2 so far?
Image can be downloaded here: https://github.com/rootzoll/raspiblitz/tree/dev#downloading-the-software

rootzoll, 2021-04-18:
Because of time issues I was not able to setup a Pihole myself for testing .. so pushing this for final testing myself to v1.7.1

PatrickScheich, 2021-04-19:
> 
> 
> @PatrickScheich @Houtek any results on this with testing v1.7RC2 so far?
> Image can be downloaded here: https://github.com/rootzoll/raspiblitz/tree/dev#downloading-the-software

I have been a bit lazy lately
But I updated my raspiblitz yesterday, the dns request spam is perfectly gone in my environment
You can consider this test as passed

rootzoll, 2021-04-20:
Perfect :D thanks for testing .. closing issue.

openoms, 2021-06-30:
This was again reported to be an issue on v1.7 by a user running a pihole.
[image]
[image]

not sure how mempool.space get pinged, it is likely from the app

flyerbear, 2022-05-01:
I wanted to add that I am seeing this behavior too - DNS queries to both v4.ipv6-test.com and mempool.space (since installing mempool space), every 12 seconds for each.

Since mempool space is self hosted are these queries really necessary?

I'm running raspiblitz v1.7.2.

rootzoll, 2022-05-17:
TODO: When running electrs there is also a call to `internet.sh status global` which does trigger a call to v4.ipv6-test.com and gets periodically called for status information by the `00blitzInfo.sh`script:
https://github.com/rootzoll/raspiblitz/blob/78a0ae00a0581cbc1b3c54f7f2770db3f2d960c8/home.admin/config.scripts/bonus.electrs.sh#L24

closed 2022-12-15 by rootzoll