raspiblitz/raspiblitz #1910: Changing the touchscreen setting broke my btrfs filesystem

holgern, 2020-12-27:
After switching off the touchscreen setting, my rapiblitz rebooted without switching off bitcoind and lnd. The dirty reboot causes that my harddisk with btrfs filesystem is now broken.

btrfs rescue zero-log /dev/sda2 shows:
parent transid verify failed on 444560588800 wanted 94010 found 93456

I'm not able to mount with:
mount -t btrfs -o rootflags=recovery,nospace_cache,clear_cache /dev/<btrfs-device> /<mountdir>

My raspiblitz is not booting anymore and I'm not able to repair my harddrive. I need to start over.

holgern, 2020-12-27:
As I'm not able to format the 32 GB stick (I was using btrfs with raid), I think everything was caused by a failing 32 gb USB stick.
I'm not able to read and I'm not able to format my stick, so I'm thinking the USB stick broke.

The broken usb-stick in RAID mode, broke then also the btrfs on the ssd.

holgern, 2020-12-28:
As the raid is not protecting against failing usb sticks, I think it would be better to remove the raid option from raspiblitz. 
This would also reduce complexity in the scripts.

openoms, 2020-12-28:
I think the RAID option is great, but need to improve the fallback option and be able to.boot in degraded mode.

Cheap USB sticks are harmful imo, see: https://github.com/rootzoll/raspiblitz/issues/92
I am running with a Sandisk USB Pro SSD with no issues. 

Great guidance from the c-lightning team on BTRFS: https://lightning.readthedocs.io/BACKUP.html#filesystem-redundancy

holgern, 2020-12-30:
Thanks, I think the link to the guide should be added to the readme. I think I broke every by using wrong btrfs commands.

rootzoll, 2024-11-14:
Closing this issue because we consider to remove BTRFS from RaspiBlitz .. see: https://github.com/raspiblitz/raspiblitz/issues/2045#issuecomment-2471845122

closed 2024-11-14 by rootzoll