raspiblitz/raspiblitz #1914: electrs txid_limit config is disabled causing queries to get stuck

softsimon, 2020-12-31:
We are currently testing mempool explorer for raspiblitz and there is one major issue right now .. If I go to a address page with many transactions the api call to electrum to fetch the txids (blockchainScripthash_getHistory), bitcoind get stuck with 100% cpu for hours because there's no limit... the default limit is actually 100 which is more reasonable, .. setting the limit to 1000 could work also .. 

This is the default config on my raspiblitz 1.6.2
```
# allow BTC-RPC-explorer show tx-s for addresses with a history of more than 100
txid_limit = 0
```

From the romanz/electrs documentation:

> doc = "Number of transactions to lookup before returning an error, to prevent 'too popular' addresses from causing the RPC server to get stuck (0 - disable the limit)"
> default = "100"

rootzoll, 2021-01-04:
For the v1.6.3 I will change the config to `txid_limit = 1000` .. should we delay the v1.6.3 to include the new mempool explorer?

wiz, 2021-01-06:
@rootzoll What was your timeline for the release? I think we shouldn't rush Simon too much, also it will take some time to work on the new build scripts and deployment environment, etc. - a lot has changed since v1

rootzoll, 2021-01-06:
@wiz after the weekend would be my latest target

wiz, 2021-01-06:
Okay we will try to ship it this weekend

rootzoll, 2021-01-10:
@wiz keep me updated

wiz, 2021-01-11:
I've just started reviewing and testing Simon's PR to add bitcoind support https://github.com/mempool/mempool/pull/281
After merging this, I can begin working on a PR to RaspiBlitz and the other Raspberry Pi distributions
It's already Monday here so I guess we won't ship it "this weekend" 😅

rootzoll, 2021-01-12:
OK. new mempool version is merged. tests looking good :) closing issue for v1.6.3 release.

closed 2021-01-12 by rootzoll