raspiblitz/raspiblitz #1916: setup error on raspiblitz

OrionRaycraft, 2021-01-01:
I have had zero error messages until now in my debug check i got this message

failed to start a high performance web server and a reverse proxy server

Thoughts?

rootzoll, 2021-01-01:
any context to that error message? on which step during the process? in which section of the debug check?

OrionRaycraft, 2021-01-01:
[image]

OrionRaycraft, 2021-01-01:
I tried to copy the log and received this error message

OrionRaycraft, 2021-01-01:
[image]

OrionRaycraft, 2021-01-01:
i can log into raspiblitz software

rootzoll, 2021-01-04:
OK .. if lnd.service is not found then something went very strange wrong. If you dont have any funds on there yet Iwould recommend a fresh start: format the HDD on your laptop clean and make a fresh sd card image to start with.

nyxnor, 2021-04-18:
@OrionRaycraft Did you solve this? Close this issue if affirmative.

nyxnor, 2021-05-17:
@rootzoll Please close this issue.

closed 2021-05-17 by rootzoll