raspiblitz/raspiblitz #1924: Lightning network behind Tor doesn't sync (stays at .01%)

OurSiDEWiND, 2021-01-09:
Hello guys! 

I wanted to do this mini-project of mine to setup a raspiblitz node in my home just for fun. A big part of why I wanted it was exactly to discover these problems during the setup-process and to learn by solving them.

I bought all the exact hardwares recommended in the raspiblitz getting started guide. My Raspi4 is an 8GB one. I balena etchered the image provided in the guide onto the sd card.

First I tried to sync the blockchain behind tor but the percentage counter did not move whatsoever after a few hours so I decided to redo the sd card, format the ssd and try without tor to sync the blockchain. After I did that the blockchain started to sync but the "Lightning process" was still at .01%. 

Now after a 2.5 day sync I am here:
[image]

My internet is pretty good with a 10mb/sec download usually from torrent sites.

When I decided whether to run the blockchain sync behind tor or not in the dialog box it wrote, that running the sync behind tor would add just 1 day to the download. Yet it made it not sync whatsoever and now I have the same problem with lightning.

What can be the cause of this?
How can I fix this?
If I sync lightning not from tor will my location get shown to people who might want to breach my home's security?
If I have to restart the whole setup process do I have to format the ssd with the blockchain on it?

Thank you for your time and your reply,
Wind

NilByte, 2021-01-09:
In my experience a complete installation on clear net could take 4 to 5 days so unless you are sure that sync stalled, just give it more time to catch up.

Ben10Brasileiro, 2021-01-09:
Dont worry, lightning will only start after 100% of the initial block download. Jus read the `debug` log if you want more information about whats happening.

kn1gh7h4wk, 2021-01-11:
I'm having the same issue. Mine is set behind tor, but I did the sync public. Blockchain gets to 99.9x% counts backward for a few days (down to 99.6x), then jumps back to 99.9x%. Mine has been running for over 2 weeks. I have a synchronous gigabit internet connection so should not be a network issues as there is plenty of bandwidth. The syncing got to 99% in just over 2 days and just sits there.

No activity on LND. It stays at .01%

The first time it got to 99.9x%, it prompted me to reboot, then it went back to the "Node is Synching (x)" page. I did try the double reboot I saw on another post, but that did not fix it.

Would appreciate any suggestions.

Thanks!

https://termbin.com/0w2z

Ben10Brasileiro, 2021-01-11:
@kn1gh7h4wk Saw your debug file. It´s trying to connect to a peer.
See this for connected nodes: https://github.com/rootzoll/raspiblitz/issues/1925#issuecomment-757433801
Than this to connect to a node: https://github.com/rootzoll/raspiblitz/issues/1925#issuecomment-757486320

OurSiDEWiND, 2021-01-12:
> I'm having the same issue. Mine is set behind tor, but I did the sync public. Blockchain gets to 99.9x% counts backward for a few days (down to 99.6x), then jumps back to 99.9x%. Mine has been [...]

Yup. had the same problem and managed to fix it. You need to manually connect to some nodes for some reason and then it will sync.

go to bitnodes.io (website) and choose whichever node addresses you prefer.
open up the console (ctrl+x) in your ssh and type bitcoin-cli addnode "insert the nodeadress here from bitnodes.io" onetry
Example: bitcoin-cli addnode "192.168.0.100:8333" onetry
Add 10 different nodes like this and it should sync pretty fast afterwards.

Keep in mind that if after a reboot the sync issue shows up again you will have to add manually nodes again to fix it. 

Cheers, 
Wind

rootzoll, 2021-01-12:
@OurSiDEWiND RaspiBlitz v1.6.3 has a bitcoind peer monitoring and will try to add peers automatically if detecting this situation

rootzoll, 2021-01-12:
If you had problems with LND connecting to other peers (running behind Tor) it might habe been caised by this attack on the Tor network: https://twitter.com/torproject/status/1348416870249074688?s=20 ... please report if lnd stays at 0 peers within next days.

OurSiDEWiND, 2021-01-12:
> If you had problems with LND connecting to other peers (running behind Tor) it might habe been caised by this attack on the Tor network: https://twitter.com/torproject/status/1348416870249074688?s=20 ... please report if lnd stays at 0 peers within next days.

Hi rootzoll. thanks for stopping by. The btc blockchain percentage counter stopped increasing and not the ln one. I did not use tor to update the blockchain. the ln one moved pretty fast to 100% as soon as the blockchain fully synced. I managed to make the blockchain progress again with with the node adding commands. 

[image]
This is what I currently see. I have 0 peers right now. (theres a blank space at the peer counter on the powershell but on the raspi display I see the number 0 on the left of the peers text) Is that normal?

Ben10Brasileiro, 2021-01-12:
@OurSiDEWiND These peers are only related to LND. To be fair, would be good if they included the peers connected to bitcoin too on this info screen, Im going to do it manually, but suggestion to the developers. To add peers for LND, search here https://1ml.com/,  type the following command: `lndcli connect insertpeerinfo`. Change insertpeerinfo by the peer you chose.

rootzoll, 2021-01-12:
@Ben10Brasileiro  v1.6.3 wil also display the number of bitcoin & lnd peers during sync 

@OurSiDEWiND without any peers its unlikely that the blockchain stays on 100% sync .. normally the sync screen should kicki in during that situation - it looks more like a unasual GUI problem not displaying the peers. maybe that is of the coloring. Will change color for v1.6.3 release

rootzoll, 2021-01-12:
marking this issue also for next release to take a look again

OurSiDEWiND, 2021-01-12:
Hmm well the node sync screen just kicked in again. 
[image]
I will try to connect to ln peers now.

kn1gh7h4wk, 2021-01-12:
Thanks @Ben10Brasileiro, @OurSiDEWiND and @rootzoll. Adding nodes fixed it immediately. It completed in less than 15min. I also went on and upgraded to the 1.6.3 rc2. Everything seems to be working fine now.

Thank you for your help!

Ben10Brasileiro, 2021-01-12:
@kn1gh7h4wk Well, 1.6.3 rc means Release Candidate, so it´s still a test version. Provide information you find relevant here https://github.com/rootzoll/raspiblitz/issues/1906. Just for me to now, you succeded with 1.6.3 with RPI3 or RPI4? I did with RPI4, but someone could traspass the boot RPI3

OurSiDEWiND, 2021-01-12:
I did the  #1859 now and these were my results. Its weird because I added 10 btc nodes a few hours ago, my blockchain synced and now as it seems like I didnt have any connections again.
[image]

Ben10Brasileiro, 2021-01-13:
@OurSiDEWiND Well, try to `reboot` the pi, `sudo systemctl status bitcoind`, `sudo systemctl restart bitcoind`. It should have been kept connected, I dont know how to solve this permanently, maybe it has something to do with your internet/ethernet connection, firewall block not sure. Ive been reading exact cases like yours, they are awaiting new release to solve this issue via API by dinamically adding nodes. But, for the time being, I created a very simplistic bash script (Im not a programmer, may have made mistakes, but it works) for your specific case, which all the nodes disconnect. This will restart every 5 minutes, so if you want to continue using your pi, just open other ssh terminal. If you want to stop the code: Ctrl+C
I enphasize, this isnt ideal, but the best I could do to keep your node syncing.
This will give you the estimated time when connected peers = 0, so you can better understand what happened.
You can change the nodes address if you want.

`nano ./con-test`
#!/bin/bash

#This is not made by raspiblitz developers, dont trust, verify.
#Retry connections
#Not to search for new connections.

count='bitcoin-cli getconnectioncount'
#now=$(date "+%T")
while true
do
        if [[ $count < 1 ]]
        then
                now=$(date "+%T")
                echo $now" - Amount of initial nodes:" && $count
                bitcoin-cli addnode "whwbo5df3a7cvkiz.onion:8333" "onetry"
                bitcoin-cli addnode "ceu54fumquesy7v7.onion:8333" "onetry"
                bitcoin-cli addnode "pzvaxvfhi24xjutk.onion:8333" "onetry"
                bitcoin-cli addnode "xyqnujd2j5fp6uy5.onion:8333" "onetry"
                bitcoin-cli addnode "wxce3fum2fjoz3cw.onion:8333" "onetry"
                bitcoin-cli addnode "hyaq7e74qqvcws3v.onion:8333" "onetry"
                bitcoin-cli addnode "sumellimpaoilabl.onion:8333" "onetry"
                bitcoin-cli addnode "5sckmx4yucbnp4io.onion:8333" "onetry"
                bitcoin-cli addnode "nqna44somaqyjqzc.onion:8333" "onetry"
                bitcoin-cli addnode "oefyu3bvzf2mobya.onion:8333" "onetry"
                now=$(date "+%T")
                echo $now " - Amount of final nodes: " && $count
                echo "-------------------------------------"
        else
                now=$(date "+%T")
                echo $now " - Enough connections" && $count
                echo "-------------------------------------"
        fi
sleep 300
done

`chmod +x con-test`
`./con-test`

OurSiDEWiND, 2021-01-18:
> I enphasize, this isnt ideal, but the best I could do to keep your node syncing.
> This will give you the estimated time when connected peers = 0, so you can better understand what happened.
> You can change the nodes address if you want.
Do I understand it correctly, my pc has to run with the ssh on in order for this script to keep working?

Ben10Brasileiro, 2021-01-19:
@OurSiDEWiND First, is it still disconnecting till 0? Have you upgraded to the version 1.6.3? And.... are you talking about lnd peers or btc peers or both? lnd uses v3 onion addresses, which were unstable cus ddos attack on the tor network, Last week I could just rarely get 3 lnd peers. Now its more stable and bigger. About the need to keep SSH open, I dont think so, because the command is ran on the pi, its just displayed on the ssh terminal, but you need test yourself to verify this. And if you forget this running for long periods of time, could overload, because it will not stop.
If you still cant keep the connections normally on the 1.6.3 version, or with the code above running, it´s better to ask @ rootzoll (without the space between).

OurSiDEWiND, 2021-01-23:
@Ben10Brasileiro The latest updated as it so far seems fixed the problem. my node synced successfully and it has peers on the btc network. thanks for all your support

rootzoll, 2021-03-24:
I also see good results with the latest update. So I am closing this issue - please comment or reopen if there is still need to improve on this.

closed 2021-03-24 by rootzoll