raspiblitz/raspiblitz #1925: Switched over to TOR, taking forever to now sync

Aki113s, 2021-01-09:
So I was not running on TOR for a long time, everything was fully synced and running for a long while, about a year, updated to 1.6.2 and for the first time switched over to TOR. Things synced faster so is TOR that much slower??? I know VPNs and secure networks mean a HUGE hit in speed.

Bitcoin 99.96%
LND at 99.97%

Been that way for 24 hours.

When I first launched it to switch to TOR I was fully synced it said, then rebooted, was in main menu and on the next Sync it is now stuck.  Is TOR that slow?? Is this normal for a Node thats just being swapped over to TOR?

T3chH, 2021-01-10:
I have the same problem on 1.6.2

chrcoe, 2021-01-10:
I'm curious if your bitcoind is not connecting properly? I am new to raspiblitz and thought I'd try tor out.

I'm seeing connection failures when i check the bitcoind logs:

```
$ sudo tail -f /mnt/hdd/bitcoin/debug.log
...snip...
2021-01-10T02:50:10Z ERROR: Error while reading proxy response
2021-01-10T02:50:10Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
...snip...
```
But I know my proxy is running and listening on that port as expected...

chrcoe, 2021-01-10:
Just saw this in the install log:

```
Err:3 https://deb.torproject.org/torproject.org buster InRelease
  The following signatures were invalid: EXPKEYSIG 74A941BA219EC810 deb.torproject.org archive signing key
Fetched 3,524 B in 2s (2,252 B/s)
Reading package lists... Done
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://deb.torproject.org/torproject.org buster InRelease: The following signatures were invalid: EXPKEYSIG 74A941BA219EC810 deb.torproject.org archive signing key
W: Failed to fetch https://deb.torproject.org/torproject.org/dists/buster/InRelease  The following signatures were invalid: EXPKEYSIG 74A941BA219EC810 deb.torproject.org archive signing key
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

🤷

Ben10Brasileiro, 2021-01-10:
About the connection refused, type the command `status` to see if there a connected nodes (`bitcoin-cli getconnectioncount` also work). About the invalid Tor signature, the raspiblitz developers are already discussing this in other topic https://github.com/rootzoll/raspiblitz/issues/1906#issuecomment-755294029

chrcoe, 2021-01-10:
> About the connection refused, type the command `status` to see if there a connected nodes (`bitcoin-cli getconnectioncount` also work). About the invalid Tor signature, the raspiblitz developers are already discussing this in other topic [#1906 (comment)](https://github.com/rootzoll/raspiblitz/issues/1906#issuecomment-755294029)

i am seeing that no nodes are connected to bitcoind:
```
$ bitcoin-cli getconnectioncount
0
```

I know that the tor proxy is working as I was able to curl through the proxy to verify IP is seen as a tor exit node and not my real IP.

Ben10Brasileiro, 2021-01-10:
You are seeing zero, despite you downloading almost everything, now there are no peers connected to you. Do the following: https://github.com/rootzoll/raspiblitz/issues/1859#issuecomment-742695230.
bitcoin-cli addnode "inputonionaddress.onion:8333 "onetry"

chrcoe, 2021-01-10:
> You are seeing zero, despite you downloading almost everything, now there are no peers connected to you. Do the following: [#1859 (comment)](https://github.com/rootzoll/raspiblitz/issues/1859#issuecomment-742695230).
> `bitcoin-cli addnode "inputonionaddress.onion:8333 "onetry"`

Thank you! that worked perfectly.  Thank you for the issue link as well.

rootzoll, 2021-01-10:
The v1.6.3 will have a automated monitoring and adding peers of it falls down to 0. Hope this will solve this for the future.

nyxnor, 2021-04-18:
@Aki113s Can you close this? Should be synced at this stage right :100:

closed 2021-04-19 by rootzoll