raspiblitz/raspiblitz #1937: raspiblitz 1.6.2 - NO HDD FOUND

DocBrown101, 2021-01-11:
Last login: Mon Jan 11 20:16:47 2021 from 192.168.0.20
Starting the main menu ...
***********************************************************
WARNING: NO HDD FOUND -> Shutdown, connect HDD and restart.
***********************************************************
If HDD is connected - please report the result of the following command:
sudo /home/admin/config.scripts/blitz.datadrive.sh status
***********************************************************
admin@raspberrypi:~ $ sudo /home/admin/config.scripts/blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0

# SETUP INFO
isSSD=0
hddCandidate=''
hddBytes=0
hddGigaBytes=0
hddPartitionCandidate=''

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=1
SwapExternalPath='/mnt/hdd/swapfile'

openoms, 2021-01-11:
What  parts are you using?

The recommendations are in the shopping list, ignore at your own peril.

DocBrown101, 2021-01-12:
This one here with 4 GB ram:
https://shop.fulmo.org/product/raspiblitz-1-6-metalcase-edition/

After a reboot and a longer sync overnight, it seems to work again now!

It happened so suddenly without any changes.

netcoding, 2021-01-14:
This Option"usb-storage.quirks=152d:1561:u" in file "/boot/cmdline.txt" it helps me. The "idVendor=152d, idProduct=1561" is the id of my controller ;-)

You can read this both article to the SSD problems.
* https://www.raspberrypi.org/forums/viewtopic.php?t=270410
* https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931

Cranky4u, 2021-01-15:
I had similair issues when I set up a Rpi 3b+ and 1TB. Turns out upping the power supply to 5V 3A was only a partial solution, had to increase cable diameter to reduce loses to the pi. Now gear is running. 

My 2 bobs worth, ensure pwr supply can pump out 3A and the cable diameter is in the 3~4mm region at least. A small signal wire of 1~3mm has to many losses.

nyxnor, 2021-04-18:
@DocBrown101 Please close this as it was solved.

closed 2021-04-18 by DocBrown101