raspiblitz/raspiblitz #1940: Freezing on i2c /dev entries

rootzoll, 2021-01-12:
> Openoms to Cranky4u
> @Cranky4u try booting without anything attached to the HDMI and use ssh then.

**EDIT1: The tips below do work on 1.6.3rc:
`hdmi_safe=1` (doesnt require to unplug when reboot).
`sudo raspi-config` > 5 Interfacing Options > P5 I2C >'Yes' (stayed on i2c /dev entries driver for 20 seconds and then loaded succesfully to the blitz menu)**
Tried with LAN and WAN, both worked.
Warning, the first boot I thought I had uncommented the hdmi _safe command, but monitor didnt lit up. I SHH finding the ip on the router settings and then edit the config.txt file to `hdmi_safe=1`. Also in the first boot it stopped at the i2c /dev, but when I ssh and changed to enagle i2c,  after a second reboot, both worked perfect.
If you guys need more details, just ask.

This is what I have for execution on config.txt (note: hdmi_safe=1 does 10 more things, see the official rpi site on config.txt)
hdmi_safe=1
disable_overscan=0
hdmi_force_hotplug=1
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtparam=audio=on

Running through Tor and Tor-sync

As for the LCD display, if someones bypass it through ssh and can access raspi-config and do the command i2c I talked several times in this post, then reboot, would be nice if it worked too.

---

I think I found a solution for `HDMI `and `I2C /dev entries driver` (the second one i2c is already in rapi-config menu, so not sure if it is failing for people that have chosen this setup, see more below).
Im still downloading the 1.6.3.... sorry if it´s not apropriate to do this on this thread.
Decided to download the 1.6.3, in the next days I will test it for having more accuracy on my comments, will edit if something is not compatible on 1.6.3 with what Im saying.
As I dont have an  LCD, I cannot verify that, correct me if im wrong with something, but thats what worked for me below.

---

My setup that works on boot (didnt need to disconnect anything or press ctrl+D):
HDMI plugged
FAN plugged (5V pins)
SSD with USB ADAPTER plugged (UBS 3.0) - both recommended in README
USB keyboard plugged
LAN plugged
MICRO SD plugged (boots from microSD)

---
HDMI >

Problem and research: not detecting hdmi, doesnt require the user to unplug and plug the cable or create a HDMI file.

What I tried and didnt work:
HDMI file
HDMI (command)
sudo /home/admin/config.scripts/blitz.lcd.sh hdmi on (same as above)

Solution (with this, there is NO need to unplug hdmi on boot): 
in `config.txt` should be like this (it enables a lot of requirements automatically, see https://www.raspberrypi.org/documentation/configuration/config-txt/video.md)
#uncomment if you get no picture on HDMI for a default "safe" mode
`hdmi_safe=1` (uncomment this line, it enables 10 more things for HDMI)

Comments: I had to unplug at boot and plug again before hdmi_safe=1, now I can maintan the cable attached during boot and the hdmi works. I tried creating hdmi file (didnt work), tried the command HDMI (didnt work either). The only solution that worked (besides unplug at boot, which needs to be done manually), the `hdmi_safe=1` at config.txt is the best option. I dont have a pi LCD screen, but even with the HDMI I was having these problems (I didnt start even after 30 minutes, had to turn the power off several times, even flashed 3 times), maybe because I2C refers to the pins of the fan.

---
I2C /dev entries driver >

Problem and reserch: hangs on boot, either doesnt start even with Alt+F2(problem 1) or just start after Alt+F2 or Ctrl+D (problem 2: sometimes Ctrl+D doesnt appear and even if it doesnt appear and I press Ctrl+D doesnt work somestimes, but sometime can bypass with Ctrl+D, but it´s manually done). Searched a lot of forums, official site, each one gives differents solutions

Solution:

Solution 1 (this worked for me, also tried several boots, unplugging and plugging HDMI, Ctrl+D appeared)
`sudo raspi-config`
5 Interfacing Options
P5 I2C
'Yes' (enable automatic loading of I2C on boot) (this does insert i2c-dev to modules and 'dtparam=i2c_arm=on', which is already in your default config file, but why not insist)

Solutions 2 (I tried to revert solution 1 so the i2c error would appear, so I could test the following, but now it just stay 5secs at the i2c /dev entries driver message and boots without needing to Ctrl+D. So, the following code, I didnt test)
on `config.txt`, continue with `dtparam=i2c=on` and add (I now some are default by the raspi, but, in hdmi some were default and just worked if it was mentioned on code, so Im not relying on default that is not written on file) (why persist if its already default? Its default on rpi4, but not all things are default on rpi3):
`ignore_lcd=0` (makes sure to detect LCD)
`display_default_lcd=1` (makes LCD the default display)
`disable_overscan=0`
`max_framebuffers=2`
`framebuffer_priority=0 (perfer main lcd)`

---

_Originally posted by @Ben10Brasileiro in https://github.com/rootzoll/raspiblitz/issues/1906#issuecomment-757415517_

rootzoll, 2021-01-12:
@openoms this looks like a great lead to fix the ic dev problem that is bugging some users for already some time. If you see someone having that problem, please let them try this solution ... and we should test this out as part of the initial sd card build for an upcoming release.

ChuckNorrison, 2021-01-16:
raspi pi 4 is a bunch of crap hardware... dont waste your time here... check out new hardware solutions, its not worth to get over a pi anymore

Ben10Brasileiro, 2021-02-22:
Chuck Norris (I wont mention you with @ so you dont get bothered), I understand you disappointment, I read almost all the issues that you were trying to get this right. You are straight up more tecnical than me, with years of experience, but I still want to find a method to make this right. You also contributed a lot, thanks.

I would like to keep this topic for discussion, even tough not getting right now, eventually I will lose power again and can test more. This post will be updated if I come to know more info or tests. It is important to log what doesnt work to wont lose time testing again. 

My assumptions: 
1. **device-tree** (other nodes use docker, no issues like this on their chats. it is related to i2c, see link 4 below) (but docker is not the percect solution)
2. **big file produced on the sd card** (just like the mempool cache problem that cant initilize it and nginx) (not the swapfile I mentioned below)

Research: I tried to discover how people were shutting down on telegram, if it was unplugging the cable, just typing reboot/shutdown, power outage, or the `off`/`restart` command. But few people give this information, they just say "I turned off and on".

Had several power outages (I mean a lot, 10 in a week, ups ordered to late...), as it is happening constantly, hangs on boot.
I tried different configurantions on the boot.txt and cmdline.txt, but still hangs at different parts.
There are some important boot messages (after i2c message I had to Ctrl+D):
Boot logs:
```
Feb 21 02:32:18 raspberrypi kernel: [    7.144614] random: systemd: uninitialized urandom read (16 bytes read)
Feb 21 02:32:18 raspberrypi kernel: [    7.787167] i2c /dev entries driver
Feb 21 02:32:18 raspberrypi kernel: [    9.479202] EXT4-fs (mmcblk0p2): warning: mounting fs with errors, running e2fsck is recommended
Feb 21 02:32:18 raspberrypi kernel: [    9.507005] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
```
and
```
[    4.811550] systemd[1]: /lib/systemd/system/fail2ban.service:12: PIDFile= references path below legacy directory /var/run/, updating /var/run/fail2ban/fail2ban.pid → /run/fail2ban/fail2ban.pid; please update the unit file accordingly.
[    4.927690] random: systemd: uninitialized urandom read (16 bytes read)
[    7.262183] EXT4-fs (mmcblk0p2): warning: mounting fs with errors, running e2fsck is recommended
[    7.347381] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   53.654362] Adding 2097148k swap on /mnt/hdd/swapfile.  Priority:-2 extents:38 across:4177920k SSFS
[   67.763851] FAT-fs (sdb1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```
Note: RPF changed some system modules, config.txt changed.

Related to systemd uniditalized https://github.com/systemd/systemd/issues/4167
I tried avoigind the systemd by remogin /run as I saw a contributor from systemd instructing in one post that I cant find right now, avoided systemd not being initialized, but caused problems to blitz later, not worth.

saw another post that Chuck Norris lowered the core_freq or arm_freq (couldnt find right now). He said it worked for some time but failed the next ones.

My thought is that a improper shutdown (or a long period to boot again) is making a huge unecessary file, next time I flash I will compare some files before and after this problems happens.
My debian (desktop, not the rpi) did this with an error file that filled up to 220G, was hard to boot but then I removed this from happening. Not the same type of errors, but still think that a file is being corrupted or enlarged on the that the boot configuration (not necessarily on the the files I mentioned, but what it triggers) of the microSD card, as everytime I flash the error stops, just appear again after lots of power loss.

Reading through the issues, even nothinf on the pins (fan or lcd), still stop. But still it wants to start the communication protocol i2c, so it doesnt care if it is in use or not, it is default to start. My fan goes crazy when powering on after a shutdown, clear i2c problem.

Even tough has smth of hardware problems, other nodes groups doenst have this, at least I didnt find, need to check if it because of device-tree, and they use docker, which changes the concept, but still better to be reluctant to and find a better solution.

Useful links: 
1. RPI config: https://elinux.org/RPiconfig
2. Config.txt: https://www.raspberrypi.org/documentation/configuration/config-txt/README.md
3. Especifically how I2C works: https://www.teachmemicro.com/i2c-primer
4. A lot of i2c conf (see topic 3.1, others too): https://www.raspberrypi.org/documentation/configuration/device-tree.md#part4.6
5. Tried this but still stopped: https://www.artekit.eu/enabling-the-i2c-on-the-raspberry-pi/

The last line..
[image]

2G generated from the sd to the hdd, by the microSD topic 53 above (dont think this is the problem, but a consequence)
[image]

rootzoll, 2021-03-25:
@openoms I think you commented out `dtparam=i2c_arm=on` in the updated build script for the 64bit RaspiOS base image - so we want to roll with the default for the v1.7? https://github.com/rootzoll/raspiblitz/blob/37b0662beb14441895df8ca5d3db5ab35d0ddac9/build_sdcard.sh#L1102

@Ben10Brasileiro any optimization on this issue we should try for the v1.7?

Do we want to activte `hdmi_safe=1`?

openoms, 2021-03-25:
@rootzoll
I2C might be needed for the touchscreen to function. I left it off to rule it out causing the I2C/dev error, which did not come up with that image.
(never could reproduce that actually).

I think we could leave I2C on for the RC image to see if it causes any issues.

Same with hdmi_safe, that would be a welcome function.

Ben10Brasileiro, 2021-03-25:
@rootzoll I tried Raspberry OS Lite for a week, had power almost failures almost daily.

The pi took longer to boot everytime a power outage happened.

I booted with nvme ssd, sata ssd, cheap usb stick and the same microSD card model I use for blitz. None, absolute zero messages about i2c /dev entries driver. The only thing that happnened was the boot time increased significantly after the first failure. But I did NOT had to manually intervene at no time of the boot process, just turn on the pi and ssh.

I booted from the microSD with the sata ssd recommended drive attached, no problem at all, just the boot time.

Im going to try the same config.txt​ and similar configuration  on cmdline​ file​ of the blitz on the raspi lite, to see if something unwanted happens.

ChuckNorrison, 2021-03-25:
At this point, i tested some power failures and observed damaged file system as expected. This will show the Control-D or maintenance screen and gets fixed with fsck -y on the sdcard on another machine.

So i can not make sure any peaks or power outages happened. UPS got installed and the RaspiBlitz is running again on 1.6.3 here beside a start9labs embassy OS solution.

At this point: start9labs does NOT run a full node and so dont use any external USB ports. He is running on 128GB microSD only. Both nodes are stable and connected to same switch and a UPS to handle any power outtages or peaks

Ben10Brasileiro, 2021-03-30:
I´ve been testing for 2 days now, did not had the time to change configurations.

  *   3 unplugs when node is running, but no i2c problem yet. No power outage yet, just manualy unplugging to test. (v1.7.0RC1 - 30/03/2021).

Although, I think this is still sample to determine. Need to increase the sample and more people to report about this.

nyxnor, 2021-04-14:
Making another comment to update. 3 more failures (stable grid), no problem so far. Nobody on telegram reported this issue after they tested v1.7. @ChuckNorrison How is it going for you? v1.7.0RC2 already. Also, slowly moving forwards to accept different hardware options, if you take a look at the improvement of the build_sdcard.sh. Not complete, but heading in a good way.

ChuckNorrison, 2021-04-14:
I still observed CTRL+D after my last post with 1.6.3. 

After reflash and first reboot in initial setup another time. So i will replace SD card, it was a SanDisk Ultra SDXC 64GB card used. Now iam testing with SanDisk Max Endurance SDXC 32GB and 1.7 RC2.

rootzoll, 2021-04-18:
So far no problems with testing for v1.7 .. but to keep a longer eye on this for testing - keeping open until v1.7.1

nyxnor, 2021-06-01:
1. This is a duplicate of this old one https://github.com/rootzoll/raspiblitz/issues/1058
2. There was on report from @hashes4merkle about this with 1.7, but he uses 7 inch screen
3. No further reports.