raspiblitz/raspiblitz #1950: electrs won't start

kriloter, 2021-01-14:
Hello,
I've activated electrum server in menu>services, but after installation electrs won't start, here is log, thank you for any advice
https://termbin.com/j3om

admin@raspberrypi:~ $ sudo ls -al /home/electrs/electrs/target/release/
total 68
drwxr-xr-x   7 electrs electrs  4096 Jan 13 22:03 .
drwxr-xr-x   3 electrs electrs  4096 Jan 13 22:18 ..
drwxr-xr-x  60 electrs electrs  4096 Jan 13 22:03 build
-rw-r--r--   1 electrs electrs     0 Jan 13 22:03 .cargo-lock
drwxr-xr-x   2 electrs electrs 32768 Jan 13 22:18 deps
drwxr-xr-x   2 electrs electrs  4096 Jan 13 22:03 examples
drwxr-xr-x 196 electrs electrs 12288 Jan 13 22:03 .fingerprint
drwxr-xr-x   2 electrs electrs  4096 Jan 13 22:03 incremental

kriloter, 2021-01-14:
update:
looks it works, it wasn't fully or properly installed first time I guess, I did uninstall and install, twice, second was succesfull and it's synchronizing now ... 🤷‍♂️
Electrum: Building Index (please wait)

nyxnor, 2021-04-18:
@kriloter Pls close this.

closed 2021-04-19 by kriloter