raspiblitz/raspiblitz #1952: Bitcoin IP shown in yellow

N00bl0rd999, 2021-01-14:
Hi,
port 8333 is reachable and open, why is my BTC IP address not turning green?

NilByte, 2021-01-14:
FAQ is a good place to find stuff. ;)

https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#why-is-my-node-address-on-the-display-yellow-not-green

N00bl0rd999, 2021-01-14:
> FAQ is a good place to find stuff. ;)
> 
> https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#why-is-my-node-address-on-the-display-yellow-not-green

Yeah, I'm just not talking about the LN node address.

NilByte, 2021-01-14:
It's the same explanation.

https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#why-is-my-bitcoin-ip-on-the-display-red

N00bl0rd999, 2021-01-14:
Why is it green in those screenshots?
https://github.com/rootzoll/raspiblitz/raw/v1.6/pictures/raspiblitz.jpg
https://github.com/rootzoll/raspiblitz/raw/v1.6/pictures/ssh9z-ready.png

Is that because of them being an older version?

NilByte, 2021-01-14:
It depends on how your internet connection is configured by your provider.

You can do a port scan here and get more information.

https://www.ipfingerprints.com/portscan.php

In any case, yellow is fine and not indicating a problem.

N00bl0rd999, 2021-01-14:
As I said, both ports are indicated as "open" by port scanning services.

> It depends on how your internet connection is configured by your provider.
What exactly would that mean?

> In any case, yellow is fine and not indicating a problem.
Thank you!

amcitaly, 2021-01-14:
The picture that you post show the address all in green because TOR is activated. Personally if I use public IP and port 8333 open has never been green but always yellow and so the LND. If you want to see all green try run the node under TOR, as in the picture

gtx1070rig, 2021-01-24:
So there are no errors if dyndns address and lnd public:ip are yellow?

amcitaly, 2021-01-24:
Yellow is OK, there is no error. If you try test your port 8333 and 9735 you will see they are open and reachable from outside

rootzoll, 2021-01-24:
Yes Yellow means OK ... the RaspiBlitz can confirm that testing that port on your public IP gives an answere. BUT that RaspiBlitz cannot confirm that the service answering is really this bitcoin core - it could be also another service in your home network that is just responding on that port. Because the RaspiBlitz cannot know for sure - the address is just yellow - not green.

closed 2021-01-24 by rootzoll