raspiblitz/raspiblitz #1955: !!! BUILD FAILED --> PGP Verify not OK / signature(0) verify(0)

hobeal, 2021-01-15:
Hi, i need help please. Im getting this error.

Ben10Brasileiro, 2021-01-15:
Please provide more information. Check this same type of issue here https://github.com/rootzoll/raspiblitz/issues/797

openoms, 2021-01-15:
Please provide more info.
is it the lnd PGP key? 
Does manual verification of the LND binary work?

hobeal, 2021-01-15:
*** PREPARATION OF BITCOIN & Co ***
--2021-01-15 08: 49: 09-- https://bitcoin.org/laanwj-releases.asc
Solving bitcoin.org (bitcoin.org) ... 138.68.248.245
Connecting with bitcoin.org (bitcoin.org) [138.68.248.245]: 443 ... connected.
HTTP request sent, waiting for response ... 200 OK
Length: 32242 (31K) [application / octet stream]
Recording to: "laanwj-releases.asc"

laanwj-releases.asc 100% [========================================= =====>] 31.49K --.- KB / s in 0.001s

2021-01-15 08:49:09 (20.9 MB / s) - "laanwj-releases.asc" saved [32242/32242]

gpg: WARNING: no command provided. Trying to guess what you want ...
pub rsa4096 2015-06-24 [SC] [expires: 2022-02-10]
      01EA5486DE18A882D4C2684590C8019E36C2E964
uid Wladimir J. van der Laan (Bitcoin Core binary launch signing key) <laanwj@gmail.com>
gpg: key 90C8019E36C2E964: 51 unverified signatures due to missing keys
gpg: /home/admin/.gnupg/trustdb.gpg: trusted database has been created
gpg: key 90C8019E36C2E964: public key "Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>" imported
gpg: Total amount processed: 1
gpg: imported: 1
gpg: absolutely reliable keys not found
--2021-01-15 08: 49: 09-- https://bitcoin.org/bin/bitcoin-core-0.20.0/SHA256SUMS.asc
Solving bitcoin.org (bitcoin.org) ... 138.68.248.245
Connecting with bitcoin.org (bitcoin.org) [138.68.248.245]: 443 ... connected.
HTTP request sent, waiting for response ... 200 OK
Length: 1769 (1.7K) [application / octet stream]
Recording to: "SHA256SUMS.asc"

SHA256SUMS.asc 100% [=========================================== ===>] 1.73K --.- KB / s in 0s

2021-01-15 08:49:10 (30.3 MB / s) - "SHA256SUMS.asc" saved [1769/1769]

good signature (0)
correctKey (0)

!!! BUILD FAILED -> PGP Verify is not ok / signature (0) verify (0)

Ben10Brasileiro, 2021-01-30:
Been 2 weeks, have you upgraded to version 1.6.3?
Reflash the microSD card.

nyxnor, 2021-04-18:
Pls close this issue.

closed 2021-04-19 by rootzoll