raspiblitz/raspiblitz #1961: Loop not working (DeadlineExceeded desc = context deadline exceeded)

petrhejna, 2021-01-18:
Error: `Error: cannot initiate swap: rpc error: code = DeadlineExceeded desc = context deadline exceeded`

Version:
- RaspiBlitz v1.6.3

rootzoll, 2021-01-24:
Bug to be researched/fixed for next release.

brandonshope, 2021-03-04:
Same error here. 

**Environment**

Version of RTL - v0.10.1-boltz-integration-release
Version of lnd - 0.11.1-beta
Version of btcd, bitcoind, - v0.20.0
Browser & browser version - 88.0.4324.190 (Official Build) (64-bit)
Host Operating system - Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
Go Version - go1.16 linux/arm
RaspiBlitz - v1.6.3

loopd.log

2021-03-04 22:35:13.122 [INF] LOOPD: Loop in terms request received
2021-03-04 22:35:13.215 [INF] LOOPD: Loop in quote request received
2021-03-04 22:35:13.217 [INF] LOOPD: Loop in quote request received
2021-03-04 22:35:28.354 [INF] LOOPD: Loop in quote request received
2021-03-04 22:35:33.288 [INF] LOOPD: Loop in request received
2021-03-04 22:35:33.288 [INF] LOOP: Loop in 0.0025 BTC (last hop: <nil>)
2021-03-04 22:35:33.401 [INF] LOOP: Creating probe invoice 19aa2aa8447a7f69c56bf52007483706dd37dc7598ae046739fd1d777210612b
2021-03-04 22:35:33.422 [INF] LOOP: Initiating swap request at height 673172
2021-03-04 22:36:03.448 [ERR] LOOPD: Loop in: cannot initiate swap: rpc error: code = DeadlineExceeded desc = context deadline exceeded

rootzoll, 2021-03-25:
With v1.7RC1 we will have loop now as part of the "Lightning Terminal" that offers also cli .. @ShahanaFarooqui is it still possible to integrate?

nyxnor, 2021-04-18:
@ph-legacy @brandonshope Can you guys test the v1.7RC2 on the dev branch. There is the image to download and build_sdcard.sh if wanted. Also, RC3 coming probably this week.

rootzoll, 2021-09-16:
Running Loop solo is deprecated ... RTL will now use the the LIT cli. First test looked good - marking for final v1.7.1 testing.

closed 2022-05-09 by rootzoll