raspiblitz/raspiblitz #1965: BTCPayServer error 502 Bad Gateway (nginx/1.14.2)

openoms, 2021-01-20:
Happening with on a new install, couldn't reproduce on updates so far.
```
$ sudo systemctl status btcpayserver
fail:Configuration: No Database option was configured.
````

to fix run:
```
$ echo "sqlitefile=sqllite.db" | sudo tee -a /home/btcpay/.btcpayserver/Main/settings.config
```
or edit `/home/btcpay/.btcpayserver/Main/settings.config` manually.

This line seems to be not kicking in:
https://github.com/rootzoll/raspiblitz/blob/c395e3a6d438c2da16a231dc172c79e7cf3cbd22/home.admin/config.scripts/bonus.btcpayserver.sh#L476

we-do-it-lu, 2021-03-12:
No need to restart the service after having corrected this config file?

openoms, 2021-03-15:
@we-do-it-lu 
yes, use:
`sudo systemctl restart btcpayserver`

rootzoll, 2021-03-17:
PR #2043 was merged - final testing if this fixes the problem

rootzoll, 2021-03-17:
Ok fixed it on my test setup ... keeping open for final testing when closer to release.

rootzoll, 2021-04-18:
Looks good on testing for v1.7. Closing issue.

closed 2021-04-18 by rootzoll