raspiblitz/raspiblitz #1967: Do you need to do port-forwarding on the router?

tifDev, 2021-01-20:
F.e. It's needed to open-ports in the router if you want to run raspiblitz with a Node + Tor + ElectRS ?

rootzoll, 2021-01-21:
If you connect over Tor you dont need port-forwarding ... a Tor hidden service (onion-address) can run behind a NAT and is publicly reachable.

tifDev, 2021-01-24:
And what if I don't connect to Tor, I need to do port-forwarding in ISP's router? 

Also, I still need to do port-forwarding to run a Tor hidden service unless I connect to other unknown server running a Tor hidden service (correct me if I'm wrong)

rootzoll, 2021-01-24:
> And what if I don't connect to Tor, I still need to do port-forwarding in ISP's router?

You only need port-forwarding when you dont use Tor (so you use clearnet IP) AND you wanna be publical reachable to peers you never connected to before. You can also run a routing node without Tor behind a NAT with no forwarded ports .. it only means that your node needs always to reachout to the other peers (initiate the connection) and those peers need to be publicly available (have a working port forwarding or run on a hosted server). 

> Also, I still need to do port-forwarding to run a Tor hidden service unless I connect to other unknown server running a Tor hidden service (correct me if I'm wrong)

If you run Tor even connecting to other peers that are publicly reachable on clearnet will be done thru Tor (using an Tor exit node). If you run Tor you can basically forget about port-forwarding. The only downside of using Tor is that you are not publicly reachable to peers running on the clearnet ... but again, that only means that your Tor node then has to initiate the connection to those nodes and you can still establish a working channel.

tifDev, 2021-01-25:
Any tutorial about this? It would be nice to have this topic a bit more detailed and with an example.
I thought that I would only need to open-ports if I don't use TOR and if I would want allow other to connect to my node.

rootzoll, 2021-01-28:
@openoms is there a chapter on this (planned) in your "Managing a Node" tutorial?
https://openoms.gitbook.io/lightning-node-management/

openoms, 2021-01-28:
Not planning to cover clearnet methods. We are planning to go Tor by default.
Exposing your home IP and exposing ports on your router is not advised to avoid compromising privacy and security of the home network.

tifDev, 2021-01-29:
Yes Tor is a better option. I'll go digg more on this.

closed 2021-01-29 by tifDev