raspiblitz/raspiblitz #1968: Channel Opening Fees in Autopilot?

RC-RPB, 2021-01-22:
New to using lightning. Funded the raspiblitz with around 150k sats and then turned on autopilot. Two channels (totaling 30k) opened quickly at a cost of what read on RTL as 15k sats each... which seems comically steep. Current balance reads at 60k... I'm not even sure where the other 30k went. Is this because autopilot is set to make sure the channel opening transactions are in the next block, or are the high fees an inherent part of opening channels and unrelated to the mempool? In any case, my intuition... as a newcomer... is that there should be better warning/control for someone turning on "autopilot", which I generally took to mean "I'm new to this, make it work reasonably well under limited controls" and not "set $15 on fire I'm dying to open channels right now". 

My fault for not reading more, certainly.

rumesht, 2021-01-22:
First time i tried autopilot it overpaied way too much. That was a few months ago and i had a 7$fee. Mempool was not full at all.
I disabled it afterwards. This should be stated or at least explained. I agree.

rootzoll, 2021-01-24:
This is the point of the script where the autopilot gets switched on ... If you like you can open a PR to a warning/explanation dialog: 

https://github.com/rootzoll/raspiblitz/blob/c395e3a6d438c2da16a231dc172c79e7cf3cbd22/home.admin/00settingsMenuBasics.sh#L195

Or if you can just propse a warning/explanation text and I can add the dialog box with OK/cancel.

nyxnor, 2021-04-18:
@RC-RPB Canc you close this issue?

closed 2021-04-18 by RC-RPB