raspiblitz/raspiblitz #1970: Test 00infoBlitz.sh - Status screen on LCD screen

Ben10Brasileiro, 2021-01-24:
Could someone that have JoinMarket on and LCD screen test this script for status screen to see if it fits? I guess so, but just to be sure. Just to be more certain before opening a pull request.
These are the changes https://github.com/Ben10Brasileiro/raspiblitz/commit/30b3eac6ac4d4bcfccbebc7fea7b3d43d6dfa866?branch=30b3eac6ac4d4bcfccbebc7fea7b3d43d6dfa866&diff=unified
This is the file https://github.com/Ben10Brasileiro/raspiblitz/blob/v1.6/home.admin/00infoBlitz.sh

Reorganized the status information (The objective it is to be aligned, so no lines get displayed much more larger than others.)
Separated by general info, bitcoin, lnd. (More clear this way)
Corrected the ascii art. (Should ony be commas and right turned this  ´´´´´´
Added lnd scan progress. (Still limited, if its 1 blk behind, color_red and display -1blk, if its more blks, the same way. Only green if its 100%=synced_to_chain, prefered this way)
Prepared for bitcoin tor v3 addresses
I know, the btc peers feels kinda isolated, but with v3 they will get very close. Besides that, I didnt find a better place for tem
[image]
[image]

The white lines that I ofuscated the informations make it worse, but Im sure you will like when you seen in your screen :)
Just to give a clear ilustration, when bitcoin accepts v3, it will be shown like this:
[image]

rootzoll, 2021-01-24:
@openoms wanna give it a test ride?

openoms, 2021-01-25:
@Ben10Brasileiro please test this on an LCD screen too.
It uses far too many lines, my screen is 6 lines short, cutting the top.

2 lines because the LND version is overflowing (using v0.12.0.rc6) - the `|` before `peers` and some spaces are redundant 
and the wallet balance is overflowing due to multiple digits in the lnd wallet - `sats` is redundant

If we want grouping with empty lines the TorV3 address of the bitcoin node won't fit the screen in the current design.
Not sure if displaying the Tor address is any useful for LND either. 
Should just signify it is running over Tor for both.
The LND pub key and alias is sufficient to identify the node from the screen. 

Also the JSON parsing is broken so needed to comment it out, not sure if that is fixed on restart. Could just leave that as it is for now since that is not for the display but the WIP web status page.

Ben10Brasileiro, 2021-01-25:
> @Ben10Brasileiro please test this on an LCD screen too.
Unfortunately I dont have an LCD screen to test that, thats why I created this issue before pull.
> It uses far too many lines, my screen is 6 lines short, cutting the top.
Yes, 6 lines it´s a lot, but only possible to reduce that if btc and lnd address are below the bolt, and each address will be reduced by 1 line. (2)
>2 lines because the LND version is overflowing (using v0.12.0.rc6) - the | before peers and some spaces are redundant
The `|` was just to make more clear, but even removing that, I cant see even removing a line.
Will try to remove exceptional spaces.
> and the wallet balance is overflowing due to multiple digits in the lnd wallet - sats is redundant
Yes, sats are redundant, I just maintained, but no problem getting rid off.
I think `OK 100%` is also redundant, if that is removed, btc peers can be fit in the same line.

>Not sure if displaying the Tor address is any useful for LND either.
>Should just signify it is running over Tor for both.
>The LND pub key and alias is sufficient to identify the node from the screen.
Ok, if its just for LND, not focusing on btc.

Pull request with updated code
https://github.com/Ben10Brasileiro/raspiblitz/pull/2#issue-561167907

Ben10Brasileiro, 2021-01-25:
I dont know how to make a proper pull request, because it came with more commits from other files that I have edited on my fork 
:(
Changes updated wit what @openoms said above https://github.com/rootzoll/raspiblitz/pull/1979/commits/0fcb27f11273df28288e53bbc30bd06dda26a0d2

rootzoll, 2021-01-28:
I marked this for v1.7 release to take another look then (can take some weeks). @Ben10Brasileiro all latest modifications are in your branch in this file? https://github.com/Ben10Brasileiro/raspiblitz/blob/v1.6/home.admin/00infoBlitz.sh

Ben10Brasileiro, 2021-01-29:
Correct.

Ben10Brasileiro, 2021-01-31:
Ive contacted with someone that have a LCD, screen, soon I will update the code.

Ben10Brasileiro, 2021-02-01:
Now im managing to get it right without excessive commits, but here it is the changes. https://github.com/rootzoll/raspiblitz/commit/d8441d02dc9f2be5fd64cd191066a71fe460862d
This is the file https://github.com/Ben10Brasileiro/raspiblitz/blob/patch-3/home.admin/00infoBlitz.sh

First photo, replicated lnd v3 address (which is even bigget than btc v3 cus host info) to btcpub.
Second photo normal look, but btc is using v2 (I asked for another photo, stil waiting response, but he said it fitted).
Thir photo is with JM yield generator running, substitutes both addresses.

[image]
[image]
[image]

FeedMyTummy, 2021-02-02:
openoms and rootzoll I was looking into that file. I would like your thoughts. I noticed that "%" is being baked into the json. I think it may be a better separation of concerns if the data was represented as a float instead of hard coded format. Let the client format. This would make localization easier. Some countries use "." others use ",". The float would be localized and the % could be appended after. Having % baked in, would cause the client to have to drop last character convert string into float, localize, append % back.

rootzoll, 2021-02-21:
Thanks for researching this ... I will get back on this in the comming weeks.

openoms, 2021-03-15:
Apart from the JSON issue it is a good idea to bring the v3 Tor addressess to the bottom and let the JMstats take over.

While testing I found that the 
`00/00 Channels xxx sat wallet xxxx sat` line overflows with many digit wallets.
Would suggest to break that line to two

echo -e "\U26D3"
echo -e "\U26D3"

echo -e "\U2607"
echo -e "\U26A1"

rootzoll, 2021-03-25:
Because of time issues I push this to the v1.8 milestone that as a focus on UI/UX anyway

openoms, 2021-03-25:
@rootzoll bitcoin v0.21.0 has a v3 onion address which does not fit the current setup (takea an extra line pushing the top out).
It is ok for the RC1, but at least a minor modification is necessary.

rootzoll, 2021-04-08:
I removed the Bitcoin TorV3 from infoscreen and replaced it with block numbers for the v1.7 release.
Otherwise I push this issue to v1.8 where we will have a redesign of the LCD part anyway.