raspiblitz/raspiblitz #1975: Feature Request scheduled or triggered SCP channel backups

woeisme, 2021-01-25:
The backups should be able to stop LND, backup channels and restart LND without node downtime. 
With Public key setup to host this would be pretty seamless. 
Possible could be triggered on state changes with inotify or similar to backup after transactions come through the node.

rootzoll, 2021-01-28:
There is already an automated Static-Channel-Backup over SCP in raspiblitz: https://github.com/rootzoll/raspiblitz#b-scp-backup-target

To automatically backup the channel database is not advised ... because basically the backup is outdated once LND gets started again and is doing its next routing/transaction (which could be within seconds after the backup). And the using an old channel db on recovery can result in loss of funds.

The LND developers were thinking long on this ... and the best for such a "hot backup" seems to be done directly on the database level - to allow in the future to have a database mirrow somewhere. This can be a solution for setups on bigger internet servers - but not for a home node solution like RaspiBlitz. Here the best "not backup" solution seems to improve on the filesystem level ... having the lnd data directory running in a raid (for example with BTRFS) .. but thats still experimental and is not a "off-site" backup. For that the "Automated Static Channel Backups per SCP" are advised to use.

nyxnor, 2021-05-17:
@rootzoll Please close this

closed 2021-05-18 by rootzoll