raspiblitz/raspiblitz #1976: sending funds from on-chain wallet

bitsam, 2021-01-25:
I am trying to transfer SATS from my on-chain wallet to a Joinmarket wallet, and it is giving me an error that there are insufficient funds, I have lowered the value to a significant value that should be able to be transferred and it still errors out. I remember having the error before, is there a minimum balance that raspiblitz wants to retain in the on-chain wallet?

rootzoll, 2021-01-28:
@openoms can you give details on sending funds to joinmarket here?

openoms, 2021-01-29:
Not sure what is the problem. When you send to JM you just send a simple bitcoin transaction to a new address.

Make sure your funds in the LND wallet are comfirmed and consider using the onchain functions of RTL for a GUI.

bitsam, 2021-01-30:
The problem isn't Joinmarket related, I was just trying to move as much out of the LND onchain wallet as possible. I was calculating fees and making a judgement on what value I could send, however I was never able to get the remaining value as low as I possible. In the end I was able to move all but 617 SATS out of my wallet. Is there an easier way to send the full value out of the onchain wallet? I didn't want to  close all channels and shutdown my node. I just wanted to move 100% (minus fees) of the value of my onchain wallet.

closed 2021-01-30 by bitsam

openoms, 2021-01-30:
Yes, RTL has a sweep function which is available in the CLI as well: https://api.lightning.community/#sendcoins