raspiblitz/raspiblitz #1977: Let's Encrypt for my own domain?

xpsx, 2021-01-25:
hello,
I want to be able to setup let's encrypt for my own domain, that I have already pointed to my public IP address.

rootzoll, 2021-01-28:
The build in LetsEnctypt `acme.sh` script in RaspiBlitz uses DNS verfication to get a valid cert on a given domain.
https://github.com/acmesh-official/acme.sh

For this a bit different that normal verfication thru serving a file on port 80 because that port is not to be made public on the RaspiBlitz. The DNS verification is not supported by all domain hosters .. so there is no easy way at the moment to make it work with your own domain.

Can you see your domain provider in the supported list of `acme.sh`?
https://github.com/acmesh-official/acme.sh/wiki/dnsapi

Maybe you could make it work with "DNS-alias-mode"?
https://github.com/acmesh-official/acme.sh/wiki/DNS-alias-mode

nyxnor, 2021-04-18:
@xpsx Did the info above solve for you? Please close this issue if yes.

closed 2021-04-18 by xpsx