raspiblitz/raspiblitz #1978: LND Backup using Windows could use some more information

Kapot, 2021-01-25:
Hi, I think the info on how to grab the LND-RESCUE file could use abit more information.

Right now it is:
```
ON YOUR MAC & LINUX LAPTOP - RUN IN NEW TERMINAL:
scp -r 'admin@xxx.xxx.xxx.xxx:/home/admin/lnd-rescue-*.tar.gz' ./
ON WINDOWS USE:
scp -r admin@xxx.xxx.xxx.xxx:/home/admin/lnd-rescue-*.tar.gz ./
```

I tried it using the 'Run' tool in Windows 10. It keeps returning 'Permission Denied' while using the correct password. I found out it it because the default destination path './' is not working.

Got it working by changing the command to enter into windows 'Run': 
`scp -r admin@xxx.xxx.xxx.xxx:/home/admin/lnd-rescue-*.tar.gz /path/to/destination`
Note that the first dot of the destination needs to be removed.

rootzoll, 2021-01-28:
Thanks for testing on Win10 (I dont have such a machine here). The `./` part not working is a bit sad ... replacing it with ` /path/to/destination` would need further explanation. Marking this to be improved for next release. 

If anybody can find how to tell scp unter Win10 to place file into current working directory without manual replacement that would be great.

Ben10Brasileiro, 2021-01-30:
https://www.pugetsystems.com/labs/hpc/How-To-Use-SSH-Client-and-Server-on-Windows-10-1470/
`scp -r admin@192.168.0.100:/home/admin/lnd-rescue-*.tar.gz .`

@Kapot Try this, inside the folder you want to save the backup, Mouse right click > Command Prompt > PlaceTheCodeAbove

Yeah, I have to use windows programs for work... sadly cant use linux full time.

Kapot, 2021-01-30:
> https://www.pugetsystems.com/labs/hpc/How-To-Use-SSH-Client-and-Server-on-Windows-10-1470/
> `scp -r admin@192.168.0.100:/home/admin/lnd-rescue-*.tar.gz .`
>
> @Kapot Try this, inside the folder you want [...]

Trying after the weekend, will let you know. Thanks.

rootzoll, 2021-03-25:
OK I changed the SCP instruction for Windows (removed the final slash).
Will be part of v1.7RC1 ... to be tested for final release.

nyxnor, 2021-04-18:
@Kapot How was the test? Worked fine? If yes, pls close this issue.

rootzoll, 2021-04-18:
Should be fine .. but still was not able to get my hands on a windows machine to test myself. So keeping open for testing until v1.7.1 or anybody reports testing from windows machine.

Kapot, 2021-05-16:
Hey, sorry for the late comment. The rightmouse trick isnt the best, because on default the 'Command Prompt' does not show in the menu showing up. You'll need to activate it somewhere.

This still seems to be the best:

`scp -r admin@xxx.xxx.xxx.xxx:/home/admin/lnd-rescue-*.tar.gz path\to\destination`

The user only needs to add the destination, which for example should look like:

`scp -r admin@xxx.xxx.xxx.xxx:/home/admin/lnd-rescue-*.tar.gz C:\Users\Username\Downloads\`

> https://www.pugetsystems.com/labs/hpc/How-To-Use-SSH-Client-and-Server-on-Windows-10-1470/
> `scp -r admin@192.168.0.100:/home/admin/lnd-rescue-*.tar.gz .`
>
> @Kapot Try this, inside the folder you want [...]