raspiblitz/raspiblitz #1980: Discussion: higher LND routing fees by default

openoms, 2021-01-25:
Related to this conversation: https://github.com/lightningnetwork/lnd/issues/4954 would like to discuss if we should use a higher feerate for the RaspiBlitz nodes as default.
The toolset and educationl material is set up towards being a routing node and in my opinion raising fees from the bottom is a basic part of efficient liquidity management.
Having `bitcoin.feerate=100` in the lnd.conf could be a good start, but curious what others think.

openoms, 2021-01-25:
Started votes on 
Twitter: https://mobile.twitter.com/openoms/status/1353814728443785216 
and 
Mastodon: https://bitcoinhackers.org/web/statuses/105618462824347680

rootzoll, 2021-01-28:
Adding this on the list for v1.7 release ... to have final decission then.

rootzoll, 2021-03-25:
@openoms The voting result was "keep low, user adjusts 53,5 %" ... so I think there is no need to make changes at the moment, right? Pushing this to v1.7.1 milestone for now.

openoms, 2021-03-26:
Agree with roasbeef in that it is not our role to modify the default fees.

I think a good way here could be to expose the lnd.conf in a dialog editbox like: https://github.com/openoms/joininbox/blob/master/scripts/set.conf.sh

Could be in the menu SERVICES as part of the menu refactoring: https://github.com/rootzoll/raspiblitz/issues/2090#issuecomment-805595752

openoms, 2021-11-08:
The lnd.conf is editable from menu - SYSTEM since v1.7.0. Closing.

closed 2021-11-08 by openoms