raspiblitz/raspiblitz #1982: rm: cannot remove 'error.tmp': No such file or directory

crystal-way, 2021-01-27:
Hello,

After a powerfailure node is still syncing, but looks like it is stuck at 99,96 and 99,97%:
Time to time, on the bottom line i see this error: "rm: cannot remove 'error.tmp': No such file or directory". Yesterday (before powerfailure) there was no this error and i believe it is to long stuck at these percents and they are not changing at all. Tried to reboot it, but did not helped. 

Node is Syncing (48) ───────────┐
                                      │  Blockchain Progress :  99.96 %         │  
                                      │  Lightning Progress  :  99.97 %         │  
                                      │  Please wait - this can take some time  │  
                                      │  Use CTRL+c to EXIT to Terminal   

 rm: cannot remove 'error.tmp': No such file or directory

crystal-way, 2021-01-27:
Update: 

Node is Syncing (0) ───────────┐
                                      │  Blockchain Progress :  99.95 %         │  
                                      │  Lightning Progress  :  99.97 %         │  
                                      │  Please wait - this can take some time  │  
                                      │  Use CTRL+c to EXIT to Terminal

Ben10Brasileiro, 2021-01-27:
rm: cannot remove 'error.tmp' also appears to me when there is a power failure, it will desappear eventually.
Type the command `debug` on the terminal to understand about the sync progress and post it here using the link that will be genereated following the instructions on the bottom of the debug log.

crystal-way, 2021-01-27:
Well in my case i see that percentage of the blockchain is going down. It mean's my node is not syncing.
https://termbin.com/1vav

Ben10Brasileiro, 2021-01-27:
You are in the version 1.6.2, this problem was happening that could not connect to other peers.
`2021-01-27T19:59:43Z Potential stale tip detected, will try using extra outbound peer (last tip update: 21321 seconds ago)`

It was fixed on version 1.6.3
https://github.com/rootzoll/raspiblitz#downloading-the-software

mamar59, 2021-01-31:
i have this problem is lightning progress is blocked, i just saw an error in debug, how i can fix it
[image]

Ben10Brasileiro, 2021-01-31:
@mamar59 Have you unlocked your wallet?
`lncli unlock` to unlock the wallet.
`sudo systemctl start litd` to restart. 

Also it can be that it isn´t connection to any remote v3 addresses due to attacks on the tor network.
Seems like you´ve installed the Lightning Terminal, I haven´t tried that yet. Openoms referenced to you this issue cause you didn´t show this logs. If this doesnt works, mark him here with "@ openoms".

mamar59, 2021-01-31:
normally I have the autounlock, is there with this command "lncli unlock " does not work

[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

Ben10Brasileiro, 2021-01-31:
> normally I have the autounlock, is there with this command "lncli unlock " does not work
> 
> [lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

I will restart and try again, to see what happens, wait;

mamar59, 2021-01-31:
I think my problem comes from loopd how to uninstall it

Ben10Brasileiro, 2021-01-31:
I didnt have loop before installing the new lightning terminal, so im not sure.
To uninstall loop:
Menu > Services > Uncheck Loop

mamar59, 2021-01-31:
> I didnt have loop before installing the new lightning terminal, so im not sure.
> To uninstall loop:
> Menu > Services > Uncheck Loop

I have uninstalled loop and reboot is this is my node is online now, Obrigado for your help

nyxnor, 2021-04-18:
@crystal-way Can you close this issue? All good?

nyxnor, 2021-05-17:
This happened to me with 1.6.3 but not with 1.7.0
@rootzoll Please close this

closed 2021-05-18 by rootzoll