raspiblitz/raspiblitz #1984: 4 inch led display help

jdh3, 2021-01-28:
Was having trouble finding the display in the parts list. Took a chance on the 4 inch HDMI display found here.
http://www.lcdwiki.com/4inch_HDMI_Display-C#Download_Resources
Can I get some advice to get this display working?   It works fine in the Rasbian OS.  When I try to get it to work with Raspiblitz it just flashes a blue screen.  Is there a way to load the drivers after the RaspiBlitz OS is installed?  
Thanks

J
[image]

rootzoll, 2021-01-28:
> Was having trouble finding the display in the parts list.

Which country are you in? 

> Can I get some advice to get this display working?

Looks like an HDMI display ... so you need to switch the RaspBlitz from GPIO-LCD to HDMI mode. See details on this in the FAQ: https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#can-i-run-the-raspiblitz-without-a-displaylcd

Ben10Brasileiro, 2021-01-30:
If you already have an IP for the Pi, connect via ssh (find in the router settings):
`sudo nano /boot/config.txt`
uncomment the following that is on line 6:
`hdmi_safe=0`
Ctrl + X, Y, Enter (to save the file)
`sudo reboot`

If you cant ssh yet:
Plug the microSD to your other machine and edit the `config.txt` the same way described above. And then just try again the card on the pi.

Let me know if this works for you @jdh3. It works on a normal HDMI monitor for me.

pkepler, 2021-01-30:
I have the same problem.  The screen will never be touchscreen will it?  I really want that feature.  Anyone have another amazon link LCD that would work out of the box?  The one on the guide is gone.

jdh3, 2021-01-30:
I did get it working with the help above and some digging around with one problem that still remains.  I tried adding the html text file alone and it did not work but after Ben10Brasileiro made me aware of the config.txt file I started just changing random things and got it to boot with the screen on.
I you want this screen to work first add the html.txt file to the boot drive.  The code below will only work (for me) if the html file is present and then the code below is added.  Im sure some things below are turned on and don't need to be I was trying random things until I got it to work.

The code below copy and pasted should get it to work. 

# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=0

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
enable_uart=1
# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtoverlay=tft35a
#dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=1,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900

The only problem now is it hangs on the i2c prompt.  https://github.com/rootzoll/raspiblitz/issues/1940

I disconnected the HDMI and it will now will boot up and I can ssh into the machine.  Before it would not even boot up to SSH.  Im currently syncing the blockchain and if all goes as expected, I think will be able to turn the screen back on via ssh commands.  If someone has advice of the i2c hanging problems that would be great.  I would like to thank everyone that helped me.

pkepler, 2021-01-30:
I can boot and ssh in.  I currently am syncing the blockchain as well.  When you say "I disconnected the HDMI"  do you mean the little HDMI to HDMI converter plug that connects the screen to the pi board?
Did you have to install the drivers like the seller indicated on AMZ?
The Raspiblitz setup guide kind of acts like this should be plug and play.

jdh3, 2021-01-30:
Yes I removed the mini plug that converts it to the larger plug on the screen.   I had to do this to be able to ssh into the pi.  The strange thing is that I could not even ssh into the pi without tweaking the config.txt first.
After I got the screen to work while booting it would hang at i2c. Which is much better than the flashing blue screen.

After the hanging at i2c prompt..I restarted the pi and unplugged the hdmi and I could ssh into the pi like the instructions said.

After I get this synced up I think ill maybe play with the config file some more maybe can figure something out.  Would be nice if the screen worked everytime I powered it off and on again with out using commands.  I guess will see how it plays out in few days.

pkepler, 2021-01-30:
Sounds good.  I had another 3.5 screen I plugged in but it didn't come up right away.  I installed a driver for the 4 inch screen earlier so I am going to install the 3.5 driver now and see what happens.  If it doesn't come up right away I am going to just leave it alone for the blockchain to sync and try more things after that like you.

Ben10Brasileiro, 2021-01-31:
For touchscreen to work @pkepler @jdh3 try this:
https://www.youtube.com/watch?v=4Va3Imd8ByM&feature=youtu.be
http://www.lcdwiki.com/4inch_HDMI_Display-C
[image]

All info from where you bought.

Ben10Brasileiro, 2021-01-31:
> Yes I removed the mini hdmi plug that converts it to the larger humid plug on the screen. I had to do this to be able to ssh into the pi. The strange thing is that I could not even ssh into the pi without tweaking the config.txt first.
> After I got the screen to work while booting it would hang at i2c. Which is much better than the flashing blue screen.
> 
> After the hanging at i2c prompt..I restarted the pi and unplugged the hdmi and I could ssh into the pi like the instructions said.
@jdh3 
Nice, no screen problem and I2C. 
Just one thing I forgot to mention to you, you config.txt. are rewritten automatically after setting up blitz.
The hdmi safe made you access the screen right, but it was the first boot, the second boot will be rewritten (because you are still setting up the pi|), then you need to correct the `config.txt´ file again do `hdmi_safe=1`, cause it returned to default settings.
Try editing the file again and boot with the hdmi cable plugged, tell me if works fot you.

pkepler, 2021-01-31:
> Sounds good. I had another 3.5 screen I plugged in but it didn't come up right away. I installed a driver for the 4 inch screen earlier so I am going to install the 3.5 driver now and see what [...]

The 3.5 screen displays fine, although I don't know if the touchscreen part works yet.  Refocusing on 4 inch screen though, I am changing the display back and trying again. Excuse my Linux ignorance but when I look at these instructions:

They list drivers for the Raspbian or Ubuntu/Kali/Retropie.  I should only have to install for the Raspbian correct?

Ben10Brasileiro, 2021-01-31:
> They list drivers for the Raspbian or Ubuntu/Kali/Retropie. I should only have to install for the Raspbian correct?

Correct, only raspbian. But you should verify if this is the seller you bought from.

jdh3, 2021-01-31:
My update is I can SSH into RaspiBlitz, the screen will work until i2c prompt. The screen will not respond after that.  Everything else is working in the background.
Basically its headless.  I will try some more edits but the pi loves to rewrite the boot SD card almost every reboot the commands don't want to stay.

jdh3, 2021-01-31:
UPDATE
With the drivers installed from the command prompt (after you have RaspiBlitz installed) It appears to be working now.
`sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./MPI4008-show`

Thanks everyone

jdh3, 2021-02-01:
> > Was having trouble finding the display in the parts list.
> 
> Which country are you in?
USA @rootzoll

closed 2021-02-03 by jdh3

rootzoll, 2021-02-03:
> USA @rootzoll

@Animalmother9000 can you check if the LCD link is correct for USA?

pkepler, 2021-02-03:
Ben10Brasileiro-  What was your last "Build Failed" mean?  I am still downloading the blockchain.  As soon as its finished I can move forward to testing the screen.  jdh3, can we leave this open just a bit longer?

Ben10Brasileiro, 2021-02-03:
@pkepler its another issue I mentioned for a person to see this one. Its not related to your problems ok, dont worry.

Animalmother9000, 2021-02-03:
@rootzoll Have updated the link. The supply on Amazon is constantly changing. Also the merchants.