raspiblitz/raspiblitz #1986: Changes to ./XXdebugLogs.sh debug logs

Ben10Brasileiro, 2021-01-29:
Could not open a pull withou a lot of junk coming on, sorry.... pls recommend me decent videos on how to do this (pull request), the ones I watched seems not to be very helpful.
Based on this https://github.com/rootzoll/raspiblitz/pull/1947

Changes: https://github.com/Ben10Brasileiro/raspiblitz/commit/8a71586c900149e8049f1d74aa3bfc9f74afa807
This file: https://github.com/Ben10Brasileiro/raspiblitz/blob/deb-2nd/home.admin/XXdebugLogs.sh

After these changes are made or cancelled, dont know if there is a better way than deleting my fork and doing  modifications only on new branches, not on the default v1.6.

rootzoll, 2021-03-25:
Merged your PR into a interims fork of v1.6 and cherrypicked them over to the `dev` btranch to be part of v1.7RC1 release. Ready for final testing :)

Ben10Brasileiro, 2021-03-26:
Good. I will try to do a best PR next time.

nyxnor, 2021-04-18:
This is done, gotta cleanse.

closed 2021-04-18 by nyxnor