raspiblitz/raspiblitz #1987: !!! BUILD FAILED --> LND PGP Verify not OK / signature(0) verify(1)

Johan4j, 2021-01-29:
Hello. So when I finish the build_sdcard.sh it comes up with the error above. I followed: https://github.com/rootzoll/raspiblitz/blob/v1.6.1/README.md#build-the-sd-card-image

**--2021-01-29 03:05:01--  https://keybase.io/roasbeef/pgp_keys.asc
Resolving keybase.io (keybase.io)... 35.153.111.151, 52.202.90.27
Connecting to keybase.io (keybase.io)|35.153.111.151|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6900 (6
[image]
.7K) [text/plain]
Saving to: 'pgp_keys.asc'

pgp_keys.asc               100%[========================================>]   6.74K  --.-KB/s    in 0s

2021-01-29 03:05:02 (33.3 MB/s) - 'pgp_keys.asc' saved [6900/6900]

gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa4096 2019-10-13 [C]
      E4D85299674B2D31FAA1892E372CBD7633C61696
uid           Olaoluwa Osuntokun <laolu32@gmail.com>
sub   rsa4096 2019-10-13 [E] [expires: 2024-10-11]
sub   rsa4096 2019-10-13 [S] [expires: 2024-10-11]
sub   rsa4096 2019-10-13 [A] [expires: 2024-10-11]

!!! BUILD WARNING --> LND PGP author not as expected
Should contain PGP: BD599672C804AF2770869A048B80CD2BB8BD8132
PRESS ENTER to TAKE THE RISK if you think all is OK

gpg: key 372CBD7633C61696: "Olaoluwa Osuntokun <laolu32@gmail.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
goodSignature(0)
correctKey(1)

!!! BUILD FAILED --> LND PGP Verify not OK / signature(0) verify(1)**

Ben10Brasileiro, 2021-01-30:
You chose to build the image yourself, its not necessary but it is a seurity measure, as you can garantee its equal to the raspiblitz repository before boot.
The link you shared on the second line is about 1.6.1. Even though it is getting the correct file build_sdcard.sh from branch v1.6, its weird you went to an older version. So, to check the version you are using:
`cat /home/admin/_version.info | grep "codeVersion"`
Type this to verify if its 1.6.3. If it is or isnt the version 1.6.3, reflash again the microSD. (https://github.com/rootzoll/raspiblitz/blob/v1.6/build_sdcard.sh)

Johan4j, 2021-01-30:
> You chose to build the image yourself, its not necessary but it is a seurity measure, as you can garantee its equal to the raspiblitz repository before boot.
> The link you shared on the second line [...]

I had to build it my self because of an error while using the pre compiled image. Can you send an img to me that you know is working?

Johan4j, 2021-01-30:
I get the i2c issue

Ben10Brasileiro, 2021-01-30:
Im trying to understand the i2c issues. Its been on the Pi since the beggining. Some people say its power failure, other say its a pi bug never corrected.

This image is working the current one (https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-13.img.gz)
Im using this img, when there is a power failure, certainly stops at i2c and I need to enter the root password, but to recommend something for you about i2c, first give me some info.

What Pi (3 or 4)?
How do you normally turn off/ reboot the pi?
- [ ] Were there power failures recently for you?

- [ ] Do you have an LCD attached?

- [ ] Do you have a pin fan attached?

SSD attached to usb (3.0 or 2.0)?

- [ ] Powered SSD?

Johan4j, 2021-01-30:
> Im trying to understand the i2c issues. Its been on the Pi since the beggining. Some people say its power failure, other say its a pi bug never corrected.
>
> This image is working the current one [...]

I am using a Raspberry pi 3b+

No I have not have any power failures.

I do not have a LCD attached

Yes I do have a pin fan attached, but I have tried removing it, and it didnt work.

I am using a hdd. I have also tried disconnecting it but it didnt work.

Ben10Brasileiro, 2021-01-30:
Reflash for now, if it stops again, will try to solve that for you by other means.

Johan4j, 2021-01-30:
> Reflash for now, if it stops again, will try to solve that for you by other means.

I just tried reflashing it but it stopped

Ben10Brasileiro, 2021-01-30:
How do you know its i2c if you dont have lcd? Do you use hdmi?

Johan4j, 2021-01-30:
I plugged in hdmi

Johan4j, 2021-01-30:
and I cant ssh into it. I have added wifi and ssh with files.

Ben10Brasileiro, 2021-01-30:
Ok. For the hdmi. Plug the microSD to your computer/laptop. 
Open the file called `config.txt`
Edit the line hdmi_safe to look like this - NO # before the line (this will lower the resolution, but its the only workaround).
`hdmi_safe=1`
Add a file called hdmi. How do you do this? You create a text file and just remove the extension .txt

Ben10Brasileiro, 2021-01-30:
> How do you normally turn off/ reboot the pi
> You forgot this one.

Please answer....
Do you remove the power cable, do you type reboot, or other?

Johan4j, 2021-01-30:
> > How do you normally turn off/ reboot the pi
> > You forgot this one.
> 
> Please answer....
> Do you remove the power cable, do you type reboot, or other?

I dont turn of the pi. I just reflash the image everytime

Ben10Brasileiro, 2021-01-30:
> I dont turn of the pi. I just reflash the image everytime

Well, I dont understand how you do that.... How do you reflash the sdCard?

Type `XXshutdown.sh` wait for the screen to be black and the waint 10 more secs,. Turn off the power.
Do as described about hdmi above.

Johan4j, 2021-01-30:
> > I dont turn of the pi. I just reflash the image everytime
> 
> Well, I dont understand how you do that.... How do you reflash the sdCard?
> 
> Type `XXshutdown.sh` wait for the screen to be black and the waint 10 more secs,. Turn off the power.
> Do as described about hdmi above.

Ok wait

Ben10Brasileiro, 2021-01-30:
Do you unplug the microSD without turning off the power of the pi?
Please join telegram group, this thread is getting extensive.

Johan4j, 2021-01-30:
> Do you unplug the microSD without turning off the power of the pi?
> Please join telegram group, this thread is getting extensive.
> https://t.me/raspiblitz

I am not doing that.

Ben10Brasileiro, 2021-01-30:
I dont know to which line you answered, there is a clear communication problem.
Detail how do you do the reflash and how do you boot.

How I would recommend for you to do.
Download the image.
Use balena etcher to reflash the sd card with the image you just downloaded.
Unplug safely the microSD from your computer and plug it again so you can access the files.
Edit the /boot/config.txt as said above.
Add the wpa_supplicant (ethernet cable is best), hdmi and ssh file.
Unplug safely the microSD.
With the pi turned off, insert the microSD on the slot.
Turn on the power.
Try with hdmi attached and unattached (mine works attached on the boot, RPI4)

As you are using RPI3, it is not as stronger as RPI4.
on the i2c problem, it just stops for 1 minute, or does it appear Ctrl +D or root password option?
If it doenst appear any of this, Alt +F2 to bypass that, user=admin, password =raspiblitz (you will be able to change that after)
As it seems its your first time trying to use blitz, if it does bypass this, try waiting some time and let the Pi running.

Run this `blitz.stresstest.sh`

Johan4j, 2021-02-01:
I get the i2c problem, and I cant click ALT + F2 Nothing happends!

Ben10Brasileiro, 2021-02-01:
Could you screenshot or take a photo?
if it stops on i2c, try to wait 1 minute (https://github.com/rootzoll/raspiblitz/issues/1984#issuecomment-770409014)
Try `Ctrl+D` instead. Never seen a case where this doennt work.

rootzoll, 2021-03-23:
OK the build sd card got a big update for the upcomming v1.7RC1 release. The LND verification should be fixed and also lots of additional parameters to to custom build the image now. So I put this to final testing.

Johan4j, 2021-03-24:
> OK the build sd card got a big update for the upcomming v1.7RC1 release. The LND verification should be fixed and also lots of additional parameters to to custom build the image now. So I put this to final testing.
> 
> There were some other ideas/issue for the sd card build in here ... maybe some need to be extracted into single issues.

Hello. Thank you for answering! When is this update gonna be released to public?

Ben10Brasileiro, 2021-03-24:
@Johan4j https://github.com/rootzoll/raspiblitz/milestone/10
Milestone 1.7 Release
There is no exact date, but as Rootzoll said that the first release candidate is ready (v.17 RC1), I would say less than a month for 1.7 for the public.

rootzoll, 2021-04-17:
OK we are getting closer to a final ... tomorrow we should have a RC3 and if nothing critical pops up anymore a final v1.7 close after. Closing this issue.

closed 2021-04-17 by rootzoll