raspiblitz/raspiblitz #1988: Update Bitcoin core to v0.21.0

openoms, 2021-01-29:
Running well on a test node so far with 10 peers behind Tor.
Updated with this script:
https://github.com/openoms/bitcoin-tutorials/tree/master/raspiblitz.updates#v0210

The Tor V3 address does overflow on the display, see:  https://github.com/rootzoll/raspiblitz/issues/1970

openoms, 2021-01-31:
Connecting Bisq to a v3 onion address is not possible yet until bitcoinj supports it: https://github.com/bisq-network/bisq/issues/2873

From https://github.com/bitcoin/bitcoin/blob/master/doc/release-notes/release-notes-0.21.0.md#p2p-and-network-changes:
To use the deprecated Tor v2 service (not recommended), the onion_private_key can be copied over onion_v3_private_key, e.g. 
`cp -f onion_private_key onion_v3_private_key` 

To switch back to the deprecated v2 address on the RaspiBlitz:
```
cd /mnt/hdd/bitcoin/
sudo -u bitcoin cp -f onion_private_key onion_v3_private_key
sudo systemctl restart bitcoind
```

openoms, 2021-03-20:
we should change or at least add v3 Tor address bitcoin nodes to the bitcoin.conf

openoms, 2021-03-20:
Here is a couple my node is connected to:
```
bitcoin-cli getpeerinfo | grep '"addr":' | cut -d'"' -f 4
```

```
ira7kqcbff52wofoong2dieh2xlvmw4e7ya3znsqn7wivn6armetvrqd.onion:8333
xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333
ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333
s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333
ldvhlpsrvspquqnl3gutz7grfu5lb3m2dgnezpl3tlkxgpoiw2g5mzid.onion:8333
gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333
v3jocfnz7esxammcmffp7johg4posszv6jbhx6r7sr7kfxjcgftgdfyd.onion:8333
6necmp3ob5mb6jnprmuh54ime74uouw2rnz4s5owpizbvjo5aolrx7ad.onion:8333
2kb3xopi3vupkfmqcyd2ibdwyhnqemxgrvfjfbqbidotk3xv7comr6yd.onion:8333
```

rootzoll, 2021-03-22:
@openoms is this what you meant? https://github.com/rootzoll/raspiblitz/commit/dccb463274f981670c861a2fab771eabd41f93bb

rootzoll, 2021-04-18:
OK ... looks good so far. Closing for v1.7

closed 2021-04-18 by rootzoll