raspiblitz/raspiblitz #1989: Update LND to v0.12.0

openoms, 2021-01-29:
Working great.
Update with PGP verification: 
https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v0120-beta

There are new options to be considered default in lnd.conf to garbage collect unused invoices from the channel.db:
```
; If true, we'll attempt to garbage collect canceled invoices upon start.
gc-canceled-invoices-on-startup=true

; If true, we'll delete newly canceled invoices on the fly.
gc-canceled-invoices-on-the-fly=true
```

rootzoll, 2021-03-15:
lnd v0.12.1 is out and will be part of the new 64-bit base image

rootzoll, 2021-04-16:
OK after testing some weeks - the lnd update seems running with no problems. Closing issues.

closed 2021-04-16 by rootzoll