raspiblitz/raspiblitz #1990: Update Tor to mitigate consensus bug

openoms, 2021-01-30:
Tor is getting attacked intermittently and losing the ability to connect to v3 addresses.

Overview article: https://darknetdaily.com/?p=1463

Latest stable release (the latest binary available for ARM: ) is 0.4.4.6 https://gitweb.torproject.org/debian/tor-pristine-upstream.git/

The (?temporary) fix: https://gitlab.torproject.org/tpo/core/tor/-/issues/40237 is merged from 0.4.5

To run the latest: https://twitter.com/openoms/status/1354875652361646081 

I am running 0.4.6.0-alpha-dev on a laptop and an RPi node.
On both needed to add the tor-nightly-master repo to the sources.list as in https://2019.www.torproject.org/docs/debian.html.en#ubuntu: 
```
echo "
deb https://deb.torproject.org/torproject.org tor-nightly-master-buster main
deb-src https://deb.torproject.org/torproject.org tor-nightly-master-buster main"  | sudo tee -a  /etc/apt/sources.list
```
Then:

- on X86: `apt update && apt install tor`
- on armhf (32bit RPi) or aarch64 (64bit RPi) build from source:
menu -> UPDATE -> TOR
https://github.com/rootzoll/raspiblitz/blob/82e0d6c3714ce1b2878780c4bdef72a6417f71c7/home.admin/config.scripts/internet.tor.sh#L493

It is difficult to test if this really helps. Generally I am not losing lightning peers during the attacks, but (re)connecting to .onion services does not work.

See the current Tor consensus status on: https://consensus-health.torproject.org/

Ben10Brasileiro, 2021-03-24:
@openoms As Tor will be default on Blitz, wouldnt it be better to build Tor from source since `build_sdcard.sh`?
https://github.com/rootzoll/raspiblitz/issues/1752#issuecomment-729298150
The Blitz 1.6.3 Tor default version is 4.4.6
I know it takes time, but if the user already chose to build the sd card, build Tor from source together would be great.
Reason:
* It is not updating for armhf with `apt update`, so **NO** Tor security fixes will be done if not updated manually or with `internet.tor.sh update`
* When you deploy the image, everyone who uses it will get the latest Tor at that time. Eventually, the will have to update tor manually.

rootzoll, 2021-04-08:
@openoms do we still need to switch to the `tor-nightly` repos or is the main up-to-date enough?
https://github.com/rootzoll/raspiblitz/blob/80547080502f541105945778f79f491f1af13a79/build_sdcard.sh#L193

openoms, 2021-04-08:
stable should be fine now.
another benefit of aarch64 is the better support with up-to-date binaries

there is the UPDATE option to go to the alpha from source

closed 2021-04-08 by rootzoll