raspiblitz/raspiblitz #1991: Use an issue template

Ben10Brasileiro, 2021-01-30:
_There are times people write 1 line and expect others to understand, so I made this template
I dont like extensive issues templates, so I made this compact one.
Inspired by mempool template https://github.com/mempool/mempool/issues/new_

## Hardware and Software

"1.6.3"
"Raspbian GNU/Linux 10 (buster)"
armv7l
Raspberry Pi 4 Model B Rev 1.1

## Issue Description

### Actual Behavior:
It will be faster to understand issues

### Expected behaviour:
To be easy to write and understand

### Steps to reproduce:
Just try

### Additional info (logs): 
Code for this: [blitz.issueTemplate.txt](https://github.com/rootzoll/raspiblitz/files/5898172/blitz.issueTemplate.txt)

## Screenshots
[image]

nyxnor, 2021-04-02:
Also, take a look at this: 
https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/issues/new/choose
[image]

rootzoll, 2021-08-24:
To cut down workload of the v1.7.1 release - pushing this organizational change to v1.8

cstenglein, 2023-07-08:
Basic templates now implemented :)

PR #3952 (2023-07-08): Add Issue templates
Closes #1991 

Would be a first draft, inspired at the examples @nyxnor  posted:

https://github.com/mempool/mempool/issues/new/choose
https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/issues/new/choose