raspiblitz/raspiblitz #1992: Raspi "whiteout" screen

tweetytoo, 2021-01-30:
I bought the RaspiBlitz v1.6 - Metal Case from fulmo. Everything worked out of the box but then -- "for no reason" -- the screen went white (just all white pixels). 

I was able to "fix it" with some commands I found on the web (lost the link :( ):

`pi restart [optional]
sudo apt-get update [optional]
sudo apt-get upgrade [optional]
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show [this reboots Pi]`

...but it happened AGAIN recently and that code didn't work, before or after I updated the OS from 1.6.2 to 1.6.3 (flashing/updating the microSD)

Any ideas?

openoms, 2021-01-30:
Is the node working otherwise? CAn you connect with ssh?
The screen is white when powered, but the RPi is off.

Try a restart (unplug the LAN cable and wait until the blue light of the SSD stops flashing and replug the power).

tweetytoo, 2021-01-31:
_Is the node working otherwise?_ Yes.

_CAn you connect with ssh?_ Yes.

_The screen is white when powered, but the RPi is off._ I don't understand this.

_Try a restart (unplug the LAN cable and wait until the blue light of the SSD stops flashing and replug the power)_

I shutdown from the RPi menu, unplugged the LAN and power and then plugged in them back in. On restart, the screen was ONLY white (no boot text). Nothing has changed. I can SSH into the RPi but the screen is still white.

openoms, 2021-01-31:
Sounds weird. Makes me think that the screen got broke, but never experienced it. I don't know what are your terms with the shop, but the metal case is compatible with the type of screen recommended in the shopping list. You could consider changing it yourself as well.

tweetytoo, 2021-02-01:
While I think it's _possibly_ a hardware problem, the fact that the screen has worked (and not) based on software commands makes me think it's something on the software side.

Is there some code I can use to turn off the screen, so it's not bright white?

I guess I can buy a new screen but that may not fix the issue?

openoms, 2021-02-01:
There is no off button on the screen. 
You can try pressing gently on the top part when the node is off to make sure all pins are connecting. Can try opening the case and put some insulation between the heatsink and the screen to make sure there is no short circuit (I use some electrical tape between the metal parts).
To rule out a software issue you can try reflashing the image to the latest v1.6.3.

Did you not run `sudo apt upgrade` by any chance? That can upgrade the kernel which can cause unforeseen issues.
can check with:
`uname -a`

The default on the image is:
```
admin@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

tweetytoo, 2021-02-02:
I can check the connectors BUT the "white screen" happened without me opening/touching the case. After I did open it (to get access to the microSD), I put it back and it worked fine... until it went white again.

I reflashed to 1.6.3 and that did not fix the issue. I DID run upgrade after that (hoping it would fix the screen) and it was still white. Thus, I have

'Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux'

I don't think any of your ideas will help, sadly (I've tried them without success 1x already) but I will try them again when I get time to tinker...

openoms, 2021-02-07:
Seems related to this: https://github.com/rootzoll/raspiblitz/issues/1436#issuecomment-766440294 and https://github.com/rootzoll/raspiblitz/issues/1985

Can you try to install this for of the display driver: https://github.com/sun-23/LCD-show ?

tweetytoo, 2021-02-13:
I can try but how do I figure out what display size I have? It's needed [for the drivers](https://github.com/sun-23/LCD-show)

openoms, 2021-02-13:
> I can try but how do I figure out what display size I have? It's needed [for the drivers](https://github.com/sun-23/LCD-show)

I tgoes by the size inches. If you have one from the shopping list the is the 3.5".

tweetytoo, 2021-02-14:
I have one from the shopping list, but WHICH 3.5"?

Is there a command to identify hardware on the unit? I don't feel like taking it apart...

**3.5” RPi Display(MPI3501):**
Driver install: sudo ./LCD35-show

**3.5” HDMI Display-B(MPI3508):**
Driver install: sudo ./MPI3508-show

**MHS-3.5” RPi Display(MHS3528):**
Driver install: sudo ./MHS35-show

**MHS-3.5” RPi Display-B(MHS35XX):**
Driver install: sudo ./MHS35B-show

openoms, 2021-02-14:
@tweetytoo 
It is the: 
3.5” RPi Display(MPI3501):
Driver install: sudo ./LCD35-show

tweetytoo, 2021-02-15:
It didn't work :(

openoms, 2021-02-15:
you might have a broken screen, I suggest you to contact the shop you ordered from and return

tweetytoo, 2021-03-04:
I got the new screen and installed it with the case cover off. It worked.

Then I swapped back in the old screen. It ALSO worked.

There was a problem with the case “bending” the screen off the contacts/pins, resulting in a white screen.

I’m not sure if there’s a “fix” except to tell people to “press down harder” if it goes white.

/ticket

closed 2021-03-04 by tweetytoo