raspiblitz/raspiblitz #1993: [New] Lightning Terminal

openoms, 2021-01-30:
Will replace the separate Loop, Pool and Faraday installations.

To test:
```
# download:
wget https://raw.githubusercontent.com/openoms/raspiblitz/new-lit/home.admin/config.scripts/bonus.lit.sh
# check:
cat bonus.lit.sh
# install:
bash bonus.lit.sh on
# start service:
sudo systemctl start litd
# info:
bash bonus.lit.sh menu
```

mamar59, 2021-01-31:
Hello openoms  I tested the new terminal last night, really great working. I rebooted my raspiblitz and since it is blocked on lightning progress and I have an error, will you have an idea, or it is normal that it is so long. thanks to you

[image]

openoms, 2021-01-31:
@mamar59 your issue i not Lightning Terminal related. Seems the same as: https://github.com/rootzoll/raspiblitz/issues/1982, please discuss there. 
Looking at the output of the logs with `debug` can help a lot with details.

Ben10Brasileiro, 2021-01-31:
UI password is presented on the logs. juts a warn to anyone who post the output to this `sudo systemctl status litd`.
Openoms, I needed to restart (useful to be in the instructions), because stopping tor stopped all other services, so starting one by one was not optimal.
@mamar59 Have you tried the UI? Try the lightning pool screen to see if you get the same thing.

Ligthing pool segment, it´s saying the token was incorrect and I needed to `listauth` to check.
[image]

using `sudo systemctl status litd`
[image]
[image]

openoms, 2021-01-31:
@Ben10Brasileiro indeed running two instances of Pool can cause this issue, so in the latest version of the script Pool and Loop is uninstalled after and the data is moved to LiT. 

The uipassword can be moved to a config file, that is to be done.

You will need to go to the command line to find the token and remove it manually. It is complicated with LiT now before we add the aliases. You can remove LiT and install Pool temporarily to use the `listauth` command and remove the token.

Then when you reinstall with the latest script it should work, but indeed all to be tested still.

openoms, 2021-01-31:
OK, switched to the config file from storing parameter in the systemd service in https://github.com/rootzoll/raspiblitz/pull/1994/commits/e45f34652659fd6dc2c2567439406ad8500db902.

Aliases are added with https://github.com/rootzoll/raspiblitz/pull/1994/commits/c7579bce2199250269c8a24bd9fde3ddd4fdeff6

A reinstall with the latest script is recommended or add the aliases:
```
source /mnt/hdd/raspiblitz.conf
  echo "
alias lit-loop=\"loop --rpcserver=localhost:8443 \\
  --tlscertpath=~/.lit/tls.cert \\	
  --macaroonpath=~/.loop/${chain}net/loop.macaroon\"
alias lit-pool=\"pool --rpcserver=localhost:8443 \\
  --tlscertpath=~/.lit/tls.cert \\	
  --macaroonpath=~/.pool/${chain}net/pool.macaroon\"
alias lit-frcli=\"frcli --rpcserver=localhost:8443 \\
  --tlscertpath=~/.lit/tls.cert \\
  --macaroonpath=~/.faraday/${chain}net/faraday.macaroon\"
" | sudo tee -a /home/lit/.bashrc
```

Sample info menu now:
[image]

Still to be done is:
- [ ] SERVICES menu integration (will remove the standalone Pool, Loop, Faraday)
- [ ] integrate in recovery
- [ ] update when password_B is changed

openoms, 2021-02-05:
When getting this error:
[image]

show the file with:
```
sudo su - lit
lit-pool listauth
```
to remove:
```
rm /home/lit/.pool/mainnet/lsat.token
```

Ben10Brasileiro, 2021-02-05:
I tried with loop and pool installed and uninstalled...
Did the steps on the first post to remove and reinstall lit too.
I have to say it worked for the first time, it was not appearing the same error, was another error that I cant see now cause its 

[image]

openoms, 2021-02-06:
@Ben10Brasileiro to gevrid of that erroneous entry uninstall lit first with:
```
bash bonus.lit.sh off
```
then reinstall:
```
bash bonus.lit.sh on
```
This is the same as reinstalling from the SERVICES menu when it is implemented.

Ben10Brasileiro, 2021-02-14:
@openoms Did not worf :(
Ive flashed the sd card and deleted all but the blockchain recently, tried the first steps to install, but was appearing the warn on pool screen even tough I didnt install pool this time. So removing the file was not appearing as there was non. Did off and on, but still same result.
I installed pool, removed the token from here `/home/pool/.pool/mainnet/lsat.token.pending`, checked that removed with ls and ON for litd and START, but still same result.
Then I found it on `/mnt/hdd/app-data/.pool/mainnet/lsat.token.pending`, deleted from here too, checked and it had deleted, but appeared seconds after.

[image]

openoms, 2021-02-14:
@Ben10Brasileiro did you follow these steps?
https://github.com/rootzoll/raspiblitz/issues/1993#issuecomment-774273660 ?

what does `lit-pool listauth` show?

Ben10Brasileiro, 2021-02-14:
@openoms Yes, I had tried that before
[image]

openoms, 2021-02-14:
>Ive flashed the sd card and deleted all but the blockchain recently, tried the first steps to install, but was appearing the warn on pool screen even tough I didnt install pool this time. So removing the file was not appearing as there was non. Did off and on, but still same result.
I installed pool, removed the token from here /home/pool/.pool/mainnet/lsat.token.pending, checked that removed with ls and ON for litd and START, but still same result.

@Ben10Brasileiro you might need to start with a fresh SDcard again. 
To be clear LiT does include Pool so if that warning comes up you need to use `lit-pool listauth` to find the and remove the token.
DO not install install Pool on top of LiT beacuse then you have a duplicate of the service causing problems.

Ben10Brasileiro, 2021-02-15:
I did that at firs, but Im going to detail what I did for this results https://github.com/rootzoll/raspiblitz/issues/1993#issuecomment-778712622

Saving this history to remember what does not work.
1 Cleaned all but blockchain and flashed sd card.
2 Installed `lit` and tried sudo su - lit & lit-pool listauth & rm /home/lit/.pool/mainnet/lsat.token, but it was saved as lsat.token.pending, wich I deleted but appeared seconds after.
3 Remove `lit` and Installed `pool` as instructed here https://github.com/rootzoll/raspiblitz/issues/1993#issuecomment-770402950 to to listauth as pool user, found the token here `/home/pool/.pool/mainnet/lsat.token.pending`, after removing with `sudo rm -rf xxx`, it was removed, checked with `ls`, but appeared some seconds after.
4 Uninstalled `pool`, and tried again to remove the token with lit user, same result, it deleted but appeared seconds after.
5 Removing /mnt/hdd/app-data/.pool didnt work, gave more erros on the lit terminal.

Ben10Brasileiro, 2021-02-16:
I dont have nothing on Pool, altough they are talking about Loop above.
I think that with 0 sat, the listauth warning will keep appearing, need to fund first to test if it's the problem.

[image]

openoms, 2021-02-16:
Good find! Do you have anything in LN channels? I did not try on a fresh node yet.

Ben10Brasileiro, 2021-02-16:
No, nothing yet, need to understand coinjoin and coinjoin first :)

openoms, 2021-02-16:
So the takeaway is that you need to be able to pay 1 sat (so have at least one channel open) to avoid the pending lsat.token error in the Lightning Terminal.

openoms, 2021-03-16:
next, torify pool and loop:
```
--loop.server.proxy=                                        The host:port of a SOCKS proxy through which all connections to the loop server will be established over
--pool.proxy=                                               The host:port of a SOCKS proxy through which all connections to the pool server will be established over
```

rootzoll, 2021-03-18:
OK merged #1994 .. will make now the changes in the SERVICES MENU :)

rootzoll, 2021-03-18:
In branch `1993-lit` I have added LIT to the menus.

@openoms but to use the cli commands to work there seems to be macaroons missing

```
admin@raspberrypi:~ $ lit-pool
No help topic for '	'
-bash: --macaroonpath=/home/lit/.pool/mainnet/pool.macaroon: No such file or directory
admin@raspberrypi:~ $ lit-loop
No help topic for '	'
-bash: --macaroonpath=/home/lit/.loop/mainnet/loop.macaroon: No such file or directory
```

openoms, 2021-03-18:
It was meant to be used fromthe `lit` user, bit we could make the aliases work from `admin` as well.

I think switching to `lit` is cleaber and more self-expalanatory. Also consistent with `bos`, `joinmarket` etc

rootzoll, 2021-03-18:
I also get errors when operating thru lit user: 

```
lit@raspberrypi:~ $ loop terms
[loop] open /home/lit/.loop/mainnet/tls.cert: no such file or directory
```

```
lit@raspberrypi:~ $ pool accounts l
[pool] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp [::1]:12010: connect: connection refused"
```

openoms, 2021-03-18:
hmm, me too. Looking into it now.

openoms, 2021-03-18:
ok this adds the aliases correctly:
```
  echo "
alias lit-loop=\"loop --rpcserver=localhost:8443 \
  --tlscertpath=/home/lit/.lit/tls.cert \
  --macaroonpath=/home/lit/.loop/${chain}net/loop.macaroon\"
alias lit-pool=\"pool --rpcserver=localhost:8443 \
  --tlscertpath=/home/lit/.lit/tls.cert \
  --macaroonpath=/home/lit/.pool/${chain}net/pool.macaroon\"
alias lit-frcli=\"frcli --rpcserver=localhost:8443 \
  --tlscertpath=/home/lit/.lit/tls.cert \
  --macaroonpath=/home/lit/.faraday/${chain}net/faraday.macaroon\"
" | sudo tee -a /home/lit/.bashrc
```

and they are removed with the `lit` user on uninstall.

rootzoll, 2021-03-18:
I merged now the `1993-lit` branch into `dev` .. because menu integrations are ok. But I still get those errors: https://github.com/rootzoll/raspiblitz/issues/1993#issuecomment-802009180

BtcContributor, 2021-03-22:
Hi,
just tried and I cannot reach Lit from Tor onion site but only from local network.
Is this a known issue or is it something related specifically to me?

openoms, 2021-03-22:
@BtcContributor it works for me. Try restatying Tor:
`sudo systemctl restrt Tor`
and make sure you open the HTTPS:// .onion link.

rootzoll, 2021-04-19:
Looks good sofar ... Final Test: Upgrade my production node to v1.7RC3 and test LoopIn with LIT.

rootzoll, 2021-04-20:
LIT seems to run fine on my production node. Closing issue for v1.7

closed 2021-04-20 by rootzoll

maximlomans, 2022-04-30:
[frcli] unable to read tls cert (path: /home/lit/.faraday/mainnet/tls.cert): open /home/lit/.faraday/mainnet/tls.cert: no such file or directory