raspiblitz/raspiblitz #1995: Tried backing up channels but errored out and now won't boot or connect to network

idaWHALE, 2021-01-30:
I was trying to backup channels, and forget the error (pretty sure file permission or directory error) but I restarted. After restart it wouldn't connect to network. Solid ethernet lights on pi but only a few SD activity light flashes at first power.

Thought it might be issues related to shim fan so I removed the fan from pi and tried re-flashing SD with latest build. Lots of SD card activity on power now but still solid ethernet lights and not connected to network. I went and got a micro HDMI to see if there was anything but nothing there either. Blank screen. Any ideas?

Ben10Brasileiro, 2021-01-31:
Yes, some ideas.
After reboot, there is something stucking you from passing, cant now exactly what right now. If you not pass that, It wont connect to WLAN or ETH.
About the blank screen, get the sd card and insert in you computer. Open the config.txt file and uncomment the following line:
`hdmi_safe=1`.
And boot again the pi, you will know whats happening know.

idaWHALE, 2021-02-16:
No luck with the above, but after burning the image to sd card several times one of them finally booted.

closed 2021-02-16 by idaWHALE