raspiblitz/raspiblitz #1997: txindex progress

rafaelpac, 2021-02-01:
Hi!
I activated the BTC-RPC explorer option and got the expected outcome about the need for txindex.
But is has been 4 days now, and still it says

Blockchain Progress : 100.00 % 14 peers
Lightning Progress  : prepare sync
Please wait - this can take some time

Is it possible to check the progress of the txindex?

I am running a Raspi 3 B+, 500Gb SSD.

Also, I have a full blockchain on my laptop, but no txindex.
If I run the txindex on my laptop e start over with the Raspi, will the txindex be copied together with the blocks during the initial copy?

rootzoll, 2021-02-21:
On the terminal you can run `/home/admin/config.scripts/network.txindex.sh status` to get some more details on the ondex building.

nyxnor, 2021-04-18:
@rafaelpac Please close this issue as rootzoll solved above.

closed 2021-04-18 by rafaelpac