raspiblitz/raspiblitz #2000: Can not use loop.

aztenjin, 2021-02-06:
using loop from the command line gets me this:

loop@raspberrypi:~ $ loop in --amt 50000
[loop] unable to read macaroon path : open /home/loop/.loop/mainnet/loop.macaroon: no such file or directory
loop@raspberrypi:~ $

using loop from RTL gets me a 500 error

ERROR
Error Code
500
Error Message
Unknown Error
API URL
./api/lnd/loop/in/termsAndQuotes

not sure where logs are to get more details?

openoms, 2021-02-07:
can you running these (as it is fixed in https://github.com/rootzoll/raspiblitz/pull/1994/commits/cd9bdd3c629d953ac9c3c0d3c276c48dac512716):
```
sudo systemctl stop loopd
echo "# persist settings in app-data"
# move old data if present
sudo mv /home/loop/.loop /mnt/hdd/app-data/ 2>/dev/null
echo "# make sure the data directory exists"
sudo mkdir -p /mnt/hdd/app-data/.loop
echo "# symlink"
sudo rm -rf /home/loop/.loop # not a symlink.. delete it silently
sudo ln -s /mnt/hdd/app-data/.loop/ /home/loop/.loop
sudo chown loop:loop -R /mnt/hdd/app-data/.loop
sudo systemctl start loopd
```

aztenjin, 2021-02-07:
> can you running these (as it is fixed in [cd9bdd3](https://github.com/rootzoll/raspiblitz/commit/cd9bdd3c629d953ac9c3c0d3c276c48dac512716)):
> 
> ```
> sudo systemctl stop loopd
> echo "# persist settings in app-data"
> # move old data if present
> sudo mv /home/loop/.loop /mnt/hdd/app-data/ 2>/dev/null
> echo "# make sure the data directory exists"
> sudo mkdir -p /mnt/hdd/app-data/.loop
> echo "# symlink"
> sudo rm -rf /home/loop/.loop # not a symlink.. delete it silently
> sudo ln -s /mnt/hdd/app-data/.loop/ /home/loop/.loop
> sudo chown loop:loop -R /mnt/hdd/app-data/.loop
> sudo systemctl start loopd
> ```

I modified a little bit, but done

clarification, i didnt move /mnt/hdd/app-data/ to dev null

i moved it to a temp directory so i could preserve any data, after the move i followed the rest of the directions and moved the preexisting data back in which which consisted of these directories and file 'lnd  nginx  README.txt  subscriptions  thunderhub'

result should have still been the same effect you were looking for.

how ever, the result is still the same that i currently have

also, just so we are clear, i have reinstalled the whole raspiblitz twice now (preserving only blockchain)

openoms, 2021-02-07:
>clarification, i didnt move /mnt/hdd/app-data/ to dev null

`sudo mv /home/loop/.loop /mnt/hdd/app-data/ 2>/dev/null`
moves `/home/loop/.loop` to `/mnt/hdd/app-data/`, the `2>/dev/null` is just hiding the error output and redundant.

I can't reproduce the problem unfortunately, so will need to wait for someone else turn up, who has the same.

In the meantime you can try the Lightning Terminal: https://github.com/rootzoll/raspiblitz/issues/1993 where you can use Loop in the GUI and cli also.

nyxnor, 2021-04-18:
@aztenjin As it was solved, pls close this issue.

closed 2021-04-19 by rootzoll