raspiblitz/raspiblitz #2002: Sphinx: File permission problem

dennisreimann, 2021-02-09:
I've tried to establish a connection to a new contact and it failed with this error:

```
raspiblitz env[28860]: (node:28860) UnhandledPromiseRejectionWarning: Error: EACCES: permission denied, open '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/signer.macaroon'
```

Same goes for creating a group chat …

Updated Sphinx, but no luck. During that update (via the update menu) I also noticed that there are merge conflicts with the `config/app.json` files, but that's another separate story … 

Temporarily solved it by running `sudo chmod 644 /mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/signer.macaroon` to make it work, but this doesn't seem ideal. 

Versions: 
- Raspiblitz v1.6.3
- Sphinx v1.3.8

openoms, 2021-02-09:
For access to the the signer macaroon,  run this line:
`sudo /usr/sbin/usermod --append --groups lndsigner sphinxrelay`

Can add it to the script if works.

dennisreimann, 2021-02-10:
@openoms Thanks, your suggestion works! 👍

openoms, 2021-02-11:
ok should get this in  PR for the next release

mmarsalko, 2021-03-22:
fwiw, I was running into this issue, but instead with router.macaroon. I fixed it by doing a 
`sudo usermod -aG lndrouter sphinxrelay`. This group might also need to be added in the next version.

rootzoll, 2021-03-22:
OK added the mission permission gruoups ... closing issue for v1.7 release. Final testing on #2089

closed 2021-03-22 by rootzoll